sparkfun/Arduino_Apollo3 #311: how to set mbed-os mbed_app.json parameters?

jerabaul29, 2020-12-19:
I am trying to replicate the following tutorial:

https://os.mbed.com/docs/mbed-os/v5.14/tutorials/power-optimization.html

I have problem with doing the following:

```
To enable CPU statistics, which show you how much time is spent in various modes, add the following line to the target_overrides section of your mbed_app.json file:

           "platform.cpu-stats-enabled": 1
```

Can you provide some hints of where / how this should be set when programming through the Arduino IDE?

jerabaul29, 2020-12-19:
What I tried, so far with no success:

# method 1 (no success): add a mbed_app.json file

I created a mbed_app.json file at location: ```/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/mbed-os```, with content:

```
~/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/mbed-os$ cat mbed_app.json 
{
    "target_overrides": {
        "platform.cpu-stats-enabled": 1
    }
}
```

# method 2 (no success)

The documentation ( https://os.mbed.com/docs/mbed-os/v6.5/program-setup/advanced-configuration.html ) says that "The Arm Mbed OS configuration system gathers and interprets the configuration defined in the target in its target configuration, all mbed_lib.json files and the mbed_app.json file. The configuration system creates a single header file, mbed_config.h, that contains all of the defined configuration parameters converted into C preprocessor macros.". I tried to change the mbed_config.h file:

```
~/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/variants/SFE_ARTEMIS/mbed$ tail -5 mbed_config.h 
#define WSF_MS_PER_TICK                                                       1                                                                                                // defined by library:cordio
#define _RTE_                                                                                                                                                                  // defined by library:rtos
#define MBED_CPU_STATS_ENABLED                                                1

#endif
```

Any idea what I do wrong?

paulvha, 2020-12-24:
The MBED-OS is pre-compiled and added as a library.  This reduces the need for recompile and thus compile time for a sketch. Making changes to the files as you did are not used at all. You need to make a copy of the Mbed-os, apply the changes, create a new library, copy that library to the right variants folder and recompile you sketch... see https://github.com/sparkfun/Arduino_Apollo3/issues/279. I had made a short description with the steps I followed. It is posted as a zip in that issues.

Regards,
Paul

jerabaul29, 2020-12-24:
Perfect, I will look at this, many thanks for your help!

idea--list, 2021-02-11:
@jerabaul29 
Just pay attention with Mbed. The tutorial you linked was meant to work with Mbed OS V5.14.
Artemis boards are supported only as of Mbed OS V6.3
Having some experience with Mbed, chances are high that things you find for V5 will not work on V6 (actually even minor releases tend to break things..) or even worse they will run but have hidden malfunctions until you face them. Also the documentation of Mbed is really sparse, so you almost always need to end up reviewing the core code if you want to understand what is really happening.

jerabaul29, 2021-02-11:
Aaah, thanks for the tips! :) .

closed 2021-03-02 by Wenn0101