sparkfun/Arduino_Apollo3 #315: Arduino IDE keeps switching me to Ambiq Bootloader, when SVL works fine.

Marsfan, 2020-12-31:
I recently updated to the Apollo3 Boards package v2.0.3 after realizing that the board manager URL changed. 
On v2.0.3, under bootloader selection, the Sparkfun Variable Loader is marked as "Out of Order", while the Ambiq Secure Bootloader is "Recommended",

[image]

Thinking that maybe something with the move to mbed OS messed up the SVL, I used the Ambiq bootloader, however, I experienced a number of issues (code did not always load, and board would stop talking over serial), so on a whim I tried selecting the SVL, and it worked properly, with no issues. 

However, every now and then the IDE seems to switch back to using the Ambiq bootloader, and I have to change it back. 
Especially considering the fact that all of Sparkfun's docs recommend using the SVL, the "out of order" label, and switching to the Ambiq bootloader should be fixed.

jerabaul29, 2020-12-31:
I had similar problems with the Ambiq bootloader, see: https://github.com/sparkfun/Arduino_Apollo3/issues/310 . +1 for setting some defaults in the Arduino IDE.

Wenn0101, 2020-12-31:
@Marsfan Good catch.

This will be changed in the next release. The SVL _was_ messed up in the first couple releases of the v2 core, and fixed in v2.0.3. I was collecting feedback from our users on the fixes for the SVL before changing the labelling. The feedback I have received so far has been that the SVL is working great and that the ASB is problematic.

Expect this to change in the next release and thanks for the feedback.

Marsfan, 2020-12-31:
@Wenn0101 Cool. Any clue how to stop the IDE from switching settings around behind my back?

Wenn0101, 2021-01-07:
@Marsfan I don't have any great advice, but the next version is on the way!

Marsfan, 2021-01-11:
@Wenn0101 Not enough irritation to make me do that. I recently figured out how to set up build tasks using Arduino CLI and VS Code, so I will probably stick with that for now.

Wenn0101, 2021-01-20:
Fixed with [this pr](https://github.com/sparkfun/Arduino_Apollo3/pull/323), and included in latest release.

closed 2021-01-20 by Wenn0101