sparkfun/Arduino_Apollo3 #316: Board manager URL Change not documented

Marsfan, 2020-12-31:
A few days ago I realized that the Arduino Board manager URL for this package had changed, and while I was using v1.2.1 of this package, development had continued to v2.0.3. 

The old URL that I was using was
https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json

Later on in the same day, I was working on an issue that I had raised with the Arduino IRMP Library (specifically  ukw100/IRMP#34), and realized that that developer was also still using the old board manager URL. I did some looking around on Google to try to find when this change occurred, and could not find any documentation on it being changed, however all of the tutorials on Sparkfun's website now list the new URL (excluding some old versions of the tutorials, which still list the previous URL). 
This needs to be documented (hopefully in the old URL's package descriptions, so that users still using the old URL are will be able to see the change when going to install/update the Artemis package), so that all users can get updated to the latest version of this package.

Wenn0101, 2021-01-07:
I will get in contact with our documentation team and see what I can do about this!

mjaffee1997, 2021-02-14:
If the arduino board manager does not have the latest release, you can do the following: 

- Access the raw json file on Github from this repository (for whichever branch you want). I've attached some below: 
https://raw.githubusercontent.com/sparkfun/Arduino_Apollo3/v2.0.1/package_sparkfun_apollo3_index.json
https://raw.githubusercontent.com/sparkfun/Arduino_Apollo3/v2.0.4/package_sparkfun_apollo3_index.json
- In arduino, go to file -> preferences 
- Go to "Additional Board Manager URLs" add the json url
- Go to the board manager and the version that you added should appear in the board manager

oclyke, 2021-02-15:
@Marsfan your suggestion to place a notice in the old package's description is a good one. Perhaps you could make a PR into this repo [SparkFun Arduino Boards](https://github.com/sparkfun/Arduino_Boards/blob/master/IDE_Board_Manager/package_sparkfun_index.json). As you can see it is hard for users of the Apollo3 Arduino core to know that the package is hosted in that other repo - hence why it was moved into this repo for v2.

There is also documentation at these locations:
[README.md](https://github.com/sparkfun/Arduino_Apollo3/blob/master/README.MD)
[Artemis Development with Arduino](https://learn.sparkfun.com/tutorials/artemis-development-with-arduino#arduino-installation) tutorial

Marsfan, 2021-02-15:
@oclyke. Cool. I'll add some PR for those. Might be a few days as I am pretty busy right now, but I will look into it within the week.

closed 2021-03-02 by Wenn0101