sparkfun/Arduino_Apollo3 #317: Can't compile if windows 10 user name has a space in it.

ryanneve, 2021-01-04:
Apologies if this is the wrong place to bring this up. 

## Subject

Using the Arduino IDE 1.8.13 and SparkFun Apollo3 1.2.1, I'm unable to compile (verify) any code on any SparkFun Apollo board from my work pc, where my user name has a space in it. Works fine at home where my username doesn't have a space in it.

## Workbench

- Win 10 enterprise
- Arduino IDE 1.8.13
- Sparkfun Apollo3 1.2.1

## Steps to reproduce

Open File --> Examples --> Artemis Examples --> Example1_Blink
Select Tools --> Board --> SparkFun Apollo3 --> SparkFun Redboard Artemis ATP (or other board)
Press Verify button

## Expected behavior
Successful compile.

## Actual Behavior

Here's the error:

> ... "C:\\Users\\Ryan Neve\\AppData\\Local\\Temp\\arduino_build_50551\\core\\config\\variant.cpp.o"
> "C:\\Users\\Ryan" "Neve\\AppData\\Local\\Temp\\arduino_build_50551\\core\\core.a"
> "-LC:\\Users\\Ryan Neve\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\apollo3\\1.2.1/cores/arduino/am_sdk_ap3/CMSIS/ARM/Lib/ARM"
> -larm_cortexM4lf_math -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -static -Wl,--gc-sections,--entry,Reset_Handler -Wl,--start-group -lm -lc -lgcc -Wl,--end-group -fno-exceptions -nostdlib --specs=nano.specs -t -lstdc++ -lc -lnosys -lm
> arm-none-eabi-gcc: error: Neve\AppData\Local\Temp\arduino_build_50551\core\core.a: No such file or directory

If you look on the second line, it inserts two double quotes between my first and last name in the path to the core.a file.  It doesn't seem to do this in the paths of any other files, just core.a

Wenn0101, 2021-01-07:
@ryanneve Thanks for the report. I will look into this next week!

darianbjohnson, 2021-01-27:
Any updates on this? This is stopping my progress; I don't want to implement a work-around (e.g. symbolic links) if this is close to being resolved. Thanks!

Wenn0101, 2021-01-27:
Yes! This was fixed in [this commit](https://github.com/sparkfun/Arduino_Apollo3/commit/6f58672c8bfebceb9bd79b30b0879b19e3c5c1e8), and the fix will be in the [next release](https://github.com/sparkfun/Arduino_Apollo3/discussions/336).

In the mean time you can grab latest branch on the release-candidate branch in github. Or you can replace the platform.txt in you branch with the latest from the dev branch or release candidate branch.

ryanneve, 2021-02-02:
Will it be fixed in the version 1.2.x that the SparkFun OpenLog Artemis still uses? Version 2.0.x doesn't do me any good unfortunately.

ryanneve, 2021-04-30:
Any hope that this will be fixed in 1.2.x? Trying to compile on OpenLog Artemis.

Wenn0101, 2021-05-13:
AHA, I have figured it out.

Most windows users with a space in their name will still not encounter an issue if they have NTFS 8.3 shortnames enabled on their drive.

On my test account
```
"C:\Users\Test User\AppData\Local\Temp\arduino_build_50551\core\core.a"
```
was being simplified to 
```
"C:\Users\TESTUS~1\AppData\Local\Temp\arduino_build_50551\core\core.a" in the arduino build.
```

Now that I can recreate the problem. I have can verify that chaning line 93 in platform.txt to 
```
libs.all="{archive_file_path}" {libs.extra} {libs.ambiq_sdk} {libs.cmsis} {libs.core} {libs.variant} {libs.extra}
```
this will get you up and running and i'll include this in the next v1 release which is coming soon!

Wenn0101, 2021-05-13:
On a side note, unless you have these disabled for a specific reason you may benefit from ntfs8.3 shortnames to help with problems with spaces in your username.

I used this article as a guide to help me figure out your problem.
https://guyrleech.wordpress.com/2014/04/15/ntfs-8-3-short-names-solving-the-issues/

closed 2021-07-20 by Wenn0101