sparkfun/Arduino_Apollo3 #319: Allowing optional re-build of the full MBED-OS core

jerabaul29, 2021-01-05:
As discussed in a couple of issues (see https://github.com/sparkfun/Arduino_Apollo3/issues/311 and https://github.com/sparkfun/Arduino_Apollo3/issues/279 in particular), much of the static code is stored in a static library. This is convenient in general, but it also makes it more difficult to perform core MBED-OS changes (enabling some flags, tweaking build etc).

My question is: would it be possible to add a "raw MBED-OS core" option to the board variants, in which case the core MBED-OS code is recompiled against the code at a known location, so that users can just tweak things without the need to figure out the build process and link process by hand each time?

Wenn0101, 2021-01-07:
@jerabaul29 This is an interesting idea.

I'll look into this. We originally dismissed this idea because of the very large compile time that would come with this kind of solution. However, if it were simply an option or an advanced feature it could make more sense. I will look into including it as a supplemental solution. 

I will file this away as a potential enhancement and note it fore the future. Don't expect movement on this right away, I am focusing on bug fixes right now.

jerabaul29, 2021-01-07:
An 'expert feature' for doing that would be amazing :) . No hurry, no worries! :) Let me you if / when you have something in place so that I can give it a try.

Wenn0101, 2021-07-20:
Closed due to inactivity. I may create a discussion thread about this to see if there is still interest.

closed 2021-07-20 by Wenn0101