sparkfun/Arduino_Apollo3 #324: How to use watchdog?

jerabaul29, 2021-01-09:
I am trying to use the watchdog in an Mbed-OS way, but I am having problems. Any idea how to fix this?

```cpp
// inspired from https://os.mbed.com/docs/mbed-os/v6.6/apis/watchdog.html

#include "mbed.h"

const uint32_t TIMEOUT_MS = 5000;

void setup() {
  // put your setup code here, to run once:

}

int main()
{
    Serial.begin(57600);
    Serial.println(F("Start"));
    Serial.end();
    
    Watchdog &watchdog = Watchdog::get_instance();
    watchdog.start(TIMEOUT_MS);

    while (true) {
      Serial.begin(57600);

        if (true){
          Serial.print(F("reset watchdog"));
          Watchdog::get_instance().kick();
        }
        else{
          Serial.println(F("do not reset watchdog; will bite in a few iterations..."));
        }
        
        ThisThread::sleep_for(TIMEOUT_MS / 10);
        
      Serial.end();
    }
}
```

Does not compile:

'Watchdog' was not declared in this scope

I also tried with a:

```
    rtos::Watchdog &watchdog = rtos::Watchdog::get_instance();
```

similar to what is used for threads in the example provided, but with no success:

'Watchdog' is not a member of 'rtos'

jerabaul29, 2021-01-09:
(if I understand well, the current watchdog discussions / issues / pull requests were for the "old" Arduino core, not the MbedOS core (I need and want to use the MbedOS core).

jerabaul29, 2021-01-21:
Do you know of any workaround to get a working watchdog with the MBedOS core? :) I am deploying a set of instruments based on the Artemis in the Arctic in a couple of weeks, would sleep much better at night if they had some watchdog protection :)

adamgarbo, 2021-02-01:
Hi @jerabaul29,

I wrote the WDT library for the 1.2.1 Apollo Core and would be keen to learn more about the Mbed OS way of enabling the WDT. What version of the core are you using? I started writing the library for v2.0 but didn't get a chance to complete it yet due to some outstanding questions regarding how access to the HAL has changed.

I also conduct research in the Arctic and would be happy to chat! Feel free to drop me a line at adam.garbo (at) carleton.ca

Cheers,
Adam

jerabaul29, 2021-02-01:
Hi @adamgarbo ,

Many thanks for your answer :) .

I am using the MbedOS version that is included in the Arduino package of this repo (the Arduino package on this repo is based on an MbedOS core from the v2 and higher, so I do  not do anything "by hand").

From looking here: https://github.com/sparkfun/Arduino_Apollo3/tree/master/cores it seems that the core used is a custom fork of the MbedOS core, since the submodule points to:

https://github.com/sparkfun/mbed-os-ambiq-apollo3/tree/0c1d6206b473047fa11c621825c128e2285d151f

Can you confirm @Wenn0101 ?

I am not sure of how things work "behind the hood" but I was surprised this did not work out of the box, since this watchdog functionality is something I would have expected to be supported natively by MbedOS through some HAL calls. So not sure where there is a hickup. I wonder if it is only a question of some missing headers or declarations or something like this that prevent MbedOS from providing the functionality. Would be curious to know if you find out of it - I am too busy just now to have a look myself unfortunately...

jerabaul29, 2021-02-01:
@adamgarbo sent an email. Wrote it a bit fast, quite busy here, but looking forward to hear back from you! :)

jerabaul29, 2021-02-04:
@adamgarbo I am testing your library now :) 

Installation procedure:

- checkout your branch and implementation: https://github.com/adamgarbo/Arduino_Apollo3/tree/WDT/libraries/WDT
- copy it to: /home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries

Then things compile fine and I can upload :) .

Running your examples gives:

## Example1_WDT_basic

```
Artemis Watchdog Timer Example
Interrupt: 1 Period: 4221 ms 
Interrupt: 2 Period: 4164 ms 
Interrupt: 3 Period: 4163 ms 
Interrupt: 4 Period: 4162 ms 

++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x10003FD4, Not allowed in ISR context
Location: 0x2292F
Error Value: 0x10003FD4
Current Thread: rtx_idle Id: 0x10004148 Entry: 0x2293D StackSize: 0x200 StackMem: 0x10004490 SP: 0x1005FE9C 
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=SFE_ARTEMIS

Artemis Watchdog Timer Example
Interrupt: 1 Period: 4221 ms 
Interrupt: 2 Period: 4164 ms 
Interrupt: 3 Period: 4163 ms 
Interrupt: 4 Period: 4162 ms 

++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x10003FD4, Not allowed in ISR context
Location: 0x2292F
Error Value: 0x10003FD4
Current Thread: rtx_idle Id: 0x10004148 Entry: 0x2293D StackSize: 0x200 StackMem: 0x10004490 SP: 0x1005FE9C 
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=SFE_ARTEMIS
```

## Example2_WDT_Config

```
Artemis Watchdog Timer Example
Interrupt: 1 Period: 1033 ms
Interrupt: 2 Period: 4115 ms
Interrupt: 3 Period: 4162 ms
Interrupt: 4 Period: 4163 ms
Interrupt: 5 Period: 4162 ms
Interrupt: 6 Period: 4162 ms
Interrupt: 7 Period: 4161 ms
Interrupt: 8 Period: 4162 ms

++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x10003FD4, Not allowed in ISR context
Location: 0x229C7
Error Value: 0x10003FD4
Current Thread: rtx_idle Id: 0x10004148 Entry: 0x229D5 StackSize: 0x200 StackMem: 0x10004490 SP: 0x1005FE9C 
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=SFE_ARTEMIS
-- MbedOS Error Info --
```

## Example3_WDT_LowPower

Here I can an (unrelated I think) compilation error:

```
/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries/WDT/examples/Example3_WDT_LowPower/Example3_WDT_LowPower.ino: In function 'void goToSleep()':
Example3_WDT_LowPower:89:3: error: 'power_adc_disable' was not declared in this scope
   power_adc_disable();
   ^~~~~~~~~~~~~~~~~
/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries/WDT/examples/Example3_WDT_LowPower/Example3_WDT_LowPower.ino:89:3: note: suggested alternative: 'am_hal_adc_disable'
   power_adc_disable();
   ^~~~~~~~~~~~~~~~~
   am_hal_adc_disable
/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries/WDT/examples/Example3_WDT_LowPower/Example3_WDT_LowPower.ino: In function 'void wakeUp()':
Example3_WDT_LowPower:133:3: error: 'ap3_adc_setup' was not declared in this scope
   ap3_adc_setup();
   ^~~~~~~~~~~~~
/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries/WDT/examples/Example3_WDT_LowPower/Example3_WDT_LowPower.ino: In function 'void am_watchdog_isr()':
Example3_WDT_LowPower:152:7: error: request for member 'clear' in '(WDT_Type*)1073889280', which is of pointer type 'WDT_Type*' (maybe you meant to use '->' ?)
   WDT.clear();
       ^~~~~
Using library RTC at version 2.0.0 in folder: /home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries/RTC 
Using library WDT at version 2.0.0 in folder: /home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries/WDT 
exit status 1
'power_adc_disable' was not declared in this scope
```

jerabaul29, 2021-02-04:
So that seems to work fine, many thanks @adamgarbo , this is very helpful :) .

A couple of questions / comments:

- it looks like we get a full MbedOS error message. Is this expected at watchdog firing, or does it means that something a bit shady happens when the resetting takes place? Not that it makes any difference to me - it clearly gets the job done, but was just curious.

- This does not follow the MbedOS API, and this is a re-implementation "by hand", right? Just that I still try to wrap my head about what is provided by MbedOS on top of "native" core-M4 functionalities, vs what is done by hand and why :) .

adamgarbo, 2021-02-04:
Thanks, @jerabaul29!

This is helpful, as I'm unable to upload code to the Artemis due to being a macOS Big Sur user.

The first two examples appear to work as intended. I am unfamiliar with the "MbedOS Error Info" message. It appears to only occur when the watchdog timer forces a reset of the system. It may also be related to the WDT interrupt service routine. I'll try to look into this further.

As to your second point, both the RTC and WDT libraries are carried over from the v1.2.x core and are simply wrappers of the Ambiq HAL, not related to Mbed OS. The original intent was to reduce the complexity of the Artemis so that users could use its features without needing to use the HAL.

I'd be happy to migrate these libraries over to Mbed, but I'm still confused about what exactly "an mbed-os enabled Arduino core" entails, and whether the v2.0.x Apollo3 Core has full access to the Mbed OS API. 

Cheers,
Adam

jerabaul29, 2021-02-04:
:) 

I agree with you, that this makes the WDT much much easier to use, and it is enough for the user case I have. Just curious about what is the way to "do things right" in a sense, given that this core is going the MbedOS direction :) . But again, a huge thanks, this is super nice and great to have a WDT back to the core v2 :) .

jerabaul29, 2021-02-05:
Btw @adamgarbo this seems to work fine, would you mind opening a pull request? :)

jerabaul29, 2021-05-05:
So if I understand well the pull request from @adamgarbo has been merged in the core v2 right? :) Should we consider this solved then or is there more work you want to do before that? :) Also, should the 🤔 on the Readme for watchdog timer be updated to ✅? Or some more things needed / to come? :) .

Wenn0101, 2021-05-05:
@jerabaul29 This is fixed in the dev and release-candidate branch. It is yet to make it into a release, i'll close this once its released.

closed 2021-05-12 by Wenn0101