sparkfun/Arduino_Apollo3 #325: how to use arm_math.h

jerabaul29, 2021-01-09:
I want to use some functions in arm_math.h .

I noticed something surprising. If I do something like this:

```cpp
// with inspiration from https://os.mbed.com/users/martinsimpson/code/CMSIS_FFT_mbed_os_DAC//file/05e2c9ca68e2/main.cpp/

#include "mbed.h"
#include "arm_math.h"

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Then I get a compile error:

```
Error while detecting libraries included by /tmp/arduino_build_107529/sketch/recipe_FFT.ino.cpp
Generating function prototypes...
/home/jrlab/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/arm-none-eabi-g++ -include /home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/variants/SFE_ARTEMIS/mbed/mbed_config.h -include /home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/arduino/sdk/ArduinoSDK.h -iprefix/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/ @/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/variants/SFE_ARTEMIS/mbed/.cxx-flags -w -x c++ -E -CC -DARDUINO=10813 -DARDUINO_APOLLO3_SFE_ARTEMIS -DARDUINO_ARCH_MBED -DARDUINO_ARCH_APOLLO3 -DMBED_NO_GLOBAL_USING_DIRECTIVE -DCORDIO_ZERO_COPY_HCI -I/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/arduino -I/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/variants/SFE_ARTEMIS -I/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/arduino -I/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/arduino/mbed-bridge -I/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/arduino/mbed-bridge/core-api @/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/variants/SFE_ARTEMIS/mbed/.includes /tmp/arduino_build_107529/sketch/recipe_FFT.ino.cpp -o /tmp/arduino_build_107529/preproc/ctags_target_for_gcc_minus_e.cpp -DARDUINO_LIB_DISCOVERY_PHASE
In file included from /home/jrlab/Desktop/Git/Artemis_MbedOS_recipes/recipes/recipe_FFT/recipe_FFT.ino:6:
/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/mbed-os/cmsis/TARGET_CORTEX_M/arm_math.h:345:4: error: #error "Define according the used Cortex core ARM_MATH_CM7, ARM_MATH_CM4, ARM_MATH_CM3, ARM_MATH_CM0PLUS, ARM_MATH_CM0, ARM_MATH_ARMV8MBL, ARM_MATH_ARMV8MML"
   #error "Define according the used Cortex core ARM_MATH_CM7, ARM_MATH_CM4, ARM_MATH_CM3, ARM_MATH_CM0PLUS, ARM_MATH_CM0, ARM_MATH_ARMV8MBL, ARM_MATH_ARMV8MML"
    ^~~~~
exit status 1
Error compiling for board RedBoard Artemis.
```

The apollo3 is a cortex-M4 if I understood well, so I tried:

```cpp
// with inspiration from https://os.mbed.com/users/martinsimpson/code/CMSIS_FFT_mbed_os_DAC//file/05e2c9ca68e2/main.cpp/

#define ARM_MATH_CM4

#include "mbed.h"
#include "arm_math.h"

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

This compiles, but getting a warning:

```
Compiling sketch...
/home/jrlab/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/arm-none-eabi-g++ -include /home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/variants/SFE_ARTEMIS/mbed/mbed_config.h -include /home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/arduino/sdk/ArduinoSDK.h -iprefix/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/ @/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/variants/SFE_ARTEMIS/mbed/.cxx-flags -DARDUINO=10813 -DARDUINO_APOLLO3_SFE_ARTEMIS -DARDUINO_ARCH_MBED -DARDUINO_ARCH_APOLLO3 -DMBED_NO_GLOBAL_USING_DIRECTIVE -DCORDIO_ZERO_COPY_HCI @/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/variants/SFE_ARTEMIS/mbed/.cxx-symbols -I/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/arduino -I/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/variants/SFE_ARTEMIS -I/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/arduino -I/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/arduino/mbed-bridge -I/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/cores/arduino/mbed-bridge/core-api @/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.3/variants/SFE_ARTEMIS/mbed/.includes /tmp/arduino_build_107529/sketch/recipe_FFT.ino.cpp -o /tmp/arduino_build_107529/sketch/recipe_FFT.ino.cpp.o
/home/jrlab/Desktop/Git/Artemis_MbedOS_recipes/recipes/recipe_FFT/recipe_FFT.ino:3: warning: "ARM_MATH_CM4" redefined
 #define ARM_MATH_CM4
```

Any idea why the compiler complains about a re-definition while it was saying it was missing initialliy?

paulvha, 2021-03-29:
The issue is related to {build.variant.path}/mbed/.cxx-symbols. In here -DARM_MATH_CM4 is set.

During the "Detecting libraries used..." it will include 'defines.preproc={defines.all}', which is defined in platform.txt. However, this does not include {build.variant.path}/mbed/.cxx-symbols and math.h will fail as it is missing a definition.

When you compile now 'defines.cxx' is included which DOES have @ {build.variant.path}/mbed/.cxx-symbols. As a result it will provide a warning that it was defined already if you set the variable.

**Solution :**
In platform.txt : comment out the original defines.preproc={defines.all} (line 17) 
Then on line 22 add : defines.preproc={defines.cxx}

The problem is solved.

regards
Paul

Wenn0101, 2021-03-29:
I was just looking at this!

The above solution is great for getting things started, but there is still some functionality that does not properly link together. I am working on an update that will get this up and going.

jerabaul29, 2021-03-29:
Many thanks for looking into this :) 

A few questions:

- I understand you are mostly now developing for the core v2 based on MbedOS, but if you start looking into this, is there any hope that these functions may be made available also to the users of the core v1 for the users who are still stuck on it? :) 

- I suppose / can you confirm that this would make the DSP and similar CMSIS functions available for us users? I am particularly interested in the possibility to use the ARM FFT functions, rather than third party libraries that may not be as well optimized for the chip :) .

jerabaul29, 2021-03-29:
(and I think this is related to #326 but I never managed to get any progress on it, I am just not good enough at understanding how the core is built).

paulvha, 2021-03-29:
I think 326 is an issue with missing files, not where ARM_MATH_CM4 is defined. 

Someone long ago had the same issue. The header file is there.. but needed import of the DSP package..
https://os.mbed.com/questions/78667/Error-Undefined-symbol-arm_rfft_fast_f32/. I expect it to be the same issue here. Adding the right package to the  source code  ( e.g. from https://github.com/ARM-software/CMSIS/tree/master/CMSIS/DSP_Lib) 

With respect to making available on V1.. I am not Sparkfun.. it is down to them to decide. Personally, I do not understand the reason to maintain 2 versions  (1.2.1 and 2.0.6) It is just confusing for the users and costly for Sparkfun. 

That said I sometimes do work on V1. The resulting code is much smaller, easier to understand. OPenLog is a good implementation on V1. I assumed it had to do with Bluetooth support and hence I ported the ArduinoBLE  to V1 which I do use in my projects. 

regards,
Paul

Wenn0101, 2021-04-08:
@jerabaul29 The latest dev branch has fixes for this. I was looking at it this to get the PDM library up and running, but it likely fixed your issues as well. Can you verify if your issues were fixed or if more work is needed to support your applications? 

The fixes require a rebuild of the mbed core, so if you are trying it out, grab the latest release-candidate branch to avoid having to rebuild it yourself.

jerabaul29, 2021-04-08:
Thanks, will try it out. Very busy here lately so will see when I have time, may need to wait a bit.

jerabaul29, 2021-05-03:
@Wenn0101 great news, now I can import ```arm_math.h```. But I still cannot import some other important helper headers, for example ```#include "arm_const_structs.h"``` fails:

```
recipe_CMSIS_FFT:98:10: fatal error: arm_const_structs.h: No such file or directoryResolveLibrary(arm_const_structs.h)

 #include "arm_const_structs.h"
  -> candidates: []
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
arm_const_structs.h: No such file or directory
```

Could this be added too, and / or could the full CMSIS be made available in full? :) .

jerabaul29, 2021-05-03:
As the initial issue was solved, closing, and opening a new issue for the extra missing headers.

closed 2021-05-03 by jerabaul29