sparkfun/Arduino_Apollo3 #326: missing cmsis functions

jerabaul29, 2021-01-09:
I am trying to take some FFTs in a program. If I understand well, MbedOS uses the CMSIS hal for providing FFT functionalities. I try to do something like this:

```cpp
#define ARM_MATH_CM4  // The apollo3 is a cortex M4

#include "mbed.h"
#include "arm_math.h"

/* FFT settings */
constexpr int SAMPLES  = 512;           /* 256 real party and 256 imaginary parts */
constexpr int FFT_SIZE = SAMPLES / 2;   /* FFT size is always the same size as we have samples, so 256 in our case */
constexpr uint8_t forward_fft = 0;
constexpr uint8_t backward_fft = 0;

// Global variables for taking FFT
float32_t fft_input[SAMPLES];
float32_t fft_output[FFT_SIZE];

void setup() {

  // compute a FFT
  arm_rfft_fast_instance_f32 crrt_arm_rfft_fast_instance_f32;
  arm_rfft_fast_init_f32(&crrt_arm_rfft_fast_instance_f32, FFT_SIZE);                         // get ready
  arm_rfft_fast_f32(&crrt_arm_rfft_fast_instance_f32, fft_input, fft_output, forward_fft);    // take the FFT
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

However, I get a linker error:

```
/home/jrlab/Desktop/Git/Artemis_MbedOS_recipes/recipes/recipe_FFT/recipe_FFT.ino:21: undefined reference to `arm_rfft_fast_init_f32'
/home/jrlab/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: /home/jrlab/Desktop/Git/Artemis_MbedOS_recipes/recipes/recipe_FFT/recipe_FFT.ino:22: undefined reference to `arm_rfft_fast_f32'
```

And indeed I cannot find the implementation of these functions in the corresponding folder.

Is something missing in the core distribution / is there something I can do to be able to take these FFTs / am I trying to do this in the wrong way?

jerabaul29, 2021-01-10:
PS: I think the main repo for CMSIS is:

https://github.com/ARM-software/CMSIS_5/tree/master

And it looks like the missing functions are located here:

https://github.com/ARM-software/CMSIS_5/tree/master/CMSIS/DSP/Source/TransformFunctions

Though there are many more categories of functions available that may be useful for some other users:

https://github.com/ARM-software/CMSIS_5/tree/master/CMSIS/DSP/Source

Not sure how you may want to make these available / distribute in this repo. Also not sure if you may want to distribute directly the sources, or some compiled / optimized binaries.

Wenn0101, 2021-03-02:
@jerabaul29 Were you ever able to get this working? Have you tried making this as just an mbed example? Is the problem exclusive to Arduino?

jerabaul29, 2021-03-03:
I have given up on this, as I got no help from the CMSIS folk ("we do not support all platforms") and I got lost in the stacks of software and did not understand where / how to include it. Instead, for now, I used an ugly workaround: using kissfft. This is very suboptimal, but given the time I had, this was the best compromise.

If you would have the possibility to "add" the CMSIS functions / make them easily available to the core, and show a few examples, would be awesome. I think this would be useful to many people, as these are plenty of basic functions that are needed all the time.

jerabaul29, 2021-03-03:
I have also tried to build these by hands, but with no success. Just wrote an issue discussing my problems, here:

https://github.com/ARM-software/CMSIS_5/issues/1138

jerabaul29, 2021-03-03:
PS: one more possible workaround that I did not had time to try: using some old pre-built stuff:

https://github.com/ARM-software/CMSIS/blob/master/CMSIS/Lib/GCC/libarm_cortexM4l_math.a

as discussed in https://forums.mbed.com/t/cmsis-dsp-support-forthcoming/8465/2

But I must say I am completely confused by all of this; if you can have a look at how to get this DSP thing available on the core it would be great.

jerabaul29, 2021-03-11:
@Wenn0101 after discussing at https://github.com/ARM-software/CMSIS_5/issues/1138 it looks like:

- there is a bit of work to successfully compile all of this
- given that I have an alternative that works well enough, it is not worth for me to spend the time to get this to work

However, if you can with some help from the ARM team 1) compile the CMSIS for the MCU once and for all, 2) provide the pre-compiled binaries as a library, I think that may be really great :) .

jerabaul29, 2021-05-03:
Yeppii, not sure what the reason is, probably some fixes related to #325, but now this is working :) . Amazing, finally able to take FFTs in my programs without using external libraries :) .

Closing.

closed 2021-05-03 by jerabaul29

jerabaul29, 2021-05-03:
PS: if this is of interest, I put a small example of how to take a real FFT with the core v1.2.1 here: https://github.com/jerabaul29/Artemis_MbedOS_recipes/blob/main/recipes/recipe_CMSIS_FFT_fft_init/recipe_CMSIS_FFT_fft_init.ino . Works like a charm :) .