sparkfun/Arduino_Apollo3 #330: Problem compiling in latest release on Ubuntu: permission denied error

jerabaul29, 2021-01-22:
The Arduino IDE suggested that I update the Artemis boards. I suppose this follows the release you had a couple of days ago.

I accepted, but now getting the following kind of errors:

```
Detecting libraries used...
/home/jrmet/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/arm-none-eabi-g++ -include /home/jrmet/.arduino15/packages/SparkFun/hardware/apollo3/2.0.4/variants/SFE_ARTEMIS/mbed/mbed_config.h -include /home/jrmet/.arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/sdk/ArduinoSDK.h -iprefix/home/jrmet/.arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/ @/home/jrmet/.arduino15/packages/SparkFun/hardware/apollo3/2.0.4/variants/SFE_ARTEMIS/mbed/.cxx-flags -w -x c++ -E -CC -DARDUINO=10813 -DARDUINO_APOLLO3_SFE_ARTEMIS -DARDUINO_ARCH_MBED -DARDUINO_ARCH_APOLLO3 -DMBED_NO_GLOBAL_USING_DIRECTIVE -DCORDIO_ZERO_COPY_HCI -I/home/jrmet/.arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino -I/home/jrmet/.arduino15/packages/SparkFun/hardware/apollo3/2.0.4/variants/SFE_ARTEMIS -I/home/jrmet/.arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino -I/home/jrmet/.arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/mbed-bridge -I/home/jrmet/.arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/mbed-bridge/core-api @/home/jrmet/.arduino15/packages/SparkFun/hardware/apollo3/2.0.4/variants/SFE_ARTEMIS/mbed/.includes -I/home/jrmet/.arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/mbed-os/drivers/ /tmp/arduino_build_445418/sketch/recipe_RTC.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for opening dependency file /dev/null.d: []
<built-in>: fatal error: opening dependency file /dev/null.d: Permission deniedResolveLibrary(opening dependency file /dev/null.d)

compilation terminated.  -> candidates: []

exit status 1
Error compiling for board RedBoard Artemis.
```

This sounds strange, as /dev/null does exist on my system of course, and I am of course able to write to ir:

```
/dev$ echo "bla" > null
```

Wonder if this is because the flag makes the code try to open / "acquire"  the /dev/null file, rather than just writing to it (?).

Any idea how to fix?

Wenn0101, 2021-01-22:
I am not near a linux machine right now, but based on what changed this last build. I may have an idea that can help you out right now. In the arduino package under variants\(TARGET)\mbed\.cxx-flags try removing -MMD from this file, save, and restart arduino.

Let me know if this helps!

jerabaul29, 2021-01-22:
That did the trick, works now, perfect, many thanks :) .

Just being curious: what was this flag supposed to help with, is it one of the flags you added to speed up compilation? Should another flag be used on linux or any other way to fix? :) .

Wenn0101, 2021-01-22:
This is a flag that was added in to speed up compile times.

As for your second question, I don't know but I plan to find out!

jerabaul29, 2021-01-22:
Excellent, let me know if / when I can do some testing for you :) .

paulvha, 2021-01-24:
The -MMD is defined in the wrong place. 

As Wenn0101 indicates remove it from : variants/(TARGET)/mbed/.cxx-flags

Now in the platform.txt, around line 57 set the parameter
compiler.c.extra_flags=-MMD
compiler.cxx.extra_flags=-MMD

It will save you a 'ton of time' on the next build.

regards,
Paul

ArminJo, 2021-01-25:
I had the same error but only for SparkFun:apollo3:sfe_artemis_nano
The SparkFun:apollo3:amap3nano was compiling fine.
Please correct.
Best regards
Armin

closed 2021-03-22 by Wenn0101