sparkfun/Arduino_Apollo3 #335: help us use "effective" programming tools: provide some help / support for Artemis on PlatformIO?

jerabaul29, 2021-01-26:
Developing large projects in the Arduino IDE can quickly become quite a pain. A much better solution, which is becoming quite widespread in the community, is to use PlatformioIDE (for example as a VSCode extension).

There is a user who is working on a PlatformIO extension for supporting the Artemis boards (and is doing an amazing work, kudos, see https://github.com/nigelb/platform-apollo3blue , and specially for the MBedOS core, see the branch Core_V2 (the master branch is only working with the old core)).

However this is as far as I understand purely some "voluntary" work. Any hope that Sparkfun helps a bit there (or starts its own extension) and invests a bit in helping having a nicely working PlatformIO extension for the Artemis boards? That would greatly help make the Artemis much more pleasant to program to the end user.

idea--list, 2021-02-11:
That sounds interesting! At least one can break out of the Arduino IDE which in itself is huge!!

Wenn0101, 2021-03-02:
This would be a great enhancement. We have a open request for it [here](https://github.com/sparkfun/Arduino_Apollo3/issues/14) .

Closing this issue, to only have 1.

closed 2021-03-02 by Wenn0101