sparkfun/Arduino_Apollo3 #337: RTC low power alarm example - bricked board?

adamgarbo, 2021-02-01:
Hey @oclyke,

Just noticed that the RTC Example6_LowPower_Alarm has this line of code added:

https://github.com/sparkfun/Arduino_Apollo3/blob/58fb804e7865e76e01ef52a7691cc5f658b032d7/libraries/RTC/examples/Example6_LowPower_Alarm/Example6_LowPower_Alarm.ino#L19

Can you advise if there is code in this example that may actually lead to a bricked board? It is based on Nathan's low power code examples for the Artemis.

Cheers,
Adam

paulvha, 2021-02-26:
I have provided a detailed reason and solution for this problem on https://forum.sparkfun.com/viewtopic.php?f=169&t=55163

regards,
Paul

adamgarbo, 2021-02-26:
Thank you, @paulvha.

I'll address the changes and include them in a future PR focused on RTC/WDT functionality. I'll leave this issue open until then.

Cheers,
Adam

Wenn0101, 2021-05-12:
changes in v2.1.0. Closing to track v2 progress. PR will be pulled into v1 shortly.

closed 2021-05-12 by Wenn0101