sparkfun/Arduino_Apollo3 #340: analogRead does not read from correct pins on RedBoard Artemis

fathat, 2021-02-03:
So, what I'm finding is if I call "analogRead(A0)", I actually get back input data for... A4? Why?

The only pins I can reliably get info from is A1 and A4, and that's essentially because I guessed at their pin indexes. I have not figured out how to read from A0, A2, A3, A5. What gives? 

Correct board and COM port are selected. Believe me double checked many times. What's the proper way to read from the analog ports?

jerabaul29, 2021-02-04:
Can this be related to the issues you were mentioning about ADC @adamgarbo ?

jerabaul29, 2021-02-04:
May be related to #304 .

mjaffee1997, 2021-02-09:
I'm having the same issue as well on the artemis thing plus. Maybe analogRead() has to be called with the native pins? ie. A0 is D20 on the Thing Plus. I don't see any A0 definition in Arduino15\packages\SparkFun\hardware\apollo3\2.0.2\variants\SFE_ARTEMIS_THING_PLUS\config pins.h

adamgarbo, 2021-02-09:
It's a good question @mjaffee1997. 

While I'm familiar with the format of the v.1.2.x board definitions, I'm really not sure what they're supposed to look like in v.2.0. I took a look at all of the variant files in v.2.0 and they only really seem to have the SPI and I2C pins defined. As far as I can tell, they are missing a **_lot_** of information.

I would suggest posting on the SparkFun forums about this issue. Perhaps the board definitions have simply not yet been completed for this version of the core.

nseidle, 2021-02-09:
@mjaffee1997 - Thanks for reporting. This is the correct place to report. It's an issue with the core. We'll get it fixed shortly.

fathat, 2021-03-08:
Hi there, any news on this? Any way someone like myself (software developer with this board) could contribute to help get it fixed?

adamgarbo, 2021-03-08:
Hi @fathat,

The SparkFun team is actively working on a v2.1 release of the Apollo3 Core and I suspect that this issue will be fixed. My best guess would be that the update is at least a couple of weeks away.

Cheers,
Adam

fathat, 2021-03-08:
Thanks for the update. Looking forward to the new version. It's really a great board.

Wenn0101, 2021-03-16:
Scheduled for v2.0.6, see https://github.com/sparkfun/Arduino_Apollo3/discussions/368 to preview release.

closed 2021-03-22 by Wenn0101