sparkfun/Arduino_Apollo3 #341: Intra-library class instantiations

adamgarbo, 2021-02-04:
Hi folks,

When @oclyke rewrote the RTC library for v.2.0.x, he adopted an approach of instantiating the RTC class from directly within the library:

https://github.com/sparkfun/Arduino_Apollo3/blob/fbb2eaca27245627bf22b59f43faa232635bcd01/libraries/RTC/src/RTC.h#L59

I've rewritten the WDT library for v2.0.x, however, if I instantiate the class using `extern Apollo3WDT WDT;`, there are conflicts when writing directly to the WDT registers. Two examples of such functions are shown below: 

```
// Set number of watchdog timer ticks allowed before a watchdog interrupt event is generated
void Apollo3WDT::setInterrupt(uint8_t interrupt)
{
  WDT->CFG_b.INTVAL = interrupt;   
}

// Set number of watchdog timer ticks allowed before the watchdog will issue a system reset
void Apollo3WDT::setReset(uint8_t reset)
{
  WDT->CFG_b.RESVAL = reset; 
}
```

Is there a particular reason for instantiating the class from within the library? I'm unfamiliar if this is a standard Mbed OS approach.

One solution could be to simply rename the instantiated object to "wdt" instead of "WDT", but then I'd likely also want to rename "RTC" to "rtc" to have greater consistency on the Core. 

Cheers,
Adam

Wenn0101, 2021-03-02:
Hey,
I'll do some research into this issue. I took a look at what you have in your WDT branch, its looking pretty good.

Would you be comfortable making pull request with your changes, and we can resolve any problems in the review process.

adamgarbo, 2021-03-02:
Hi @Wenn0101,

Happy to create a PR. I assume to the release-candidate branch?

Wenn0101, 2021-03-02:
Thanks! The dev branch actually. (release-candidate is autogenerate from commits to that branch)

closed 2021-05-12 by Wenn0101