sparkfun/Arduino_Apollo3 #343: Apollo3 Core v2.0.5 - Missing Artemis MicroMod Processor

adamgarbo, 2021-02-04:
Hi folks,

A quick heads up that the Artemis MicroMod Processor Board is missing from v.2.0.5 of the Apollo3 Core. While it's selectable in the Arduino IDE, there doesn't appear to be a corresponding `TARGET_SFE_ARTEMIS_MM_PB` folder located in: `~/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/mbed-os/targets/TARGET_Ambiq_Micro/TARGET_Apollo3`.

Compilation output below:
```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/adam/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/adam/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/adam/Documents/Arduino/libraries -fqbn=SparkFun:apollo3:sfe_artemis_mm_pb:svl_baud=921600,loader=option_svl -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/hm/mbygfq0s3g1f3vpv6dg01c7w0000gn/T/arduino_build_589717 -warnings=none -build-cache /var/folders/hm/mbygfq0s3g1f3vpv6dg01c7w0000gn/T/arduino_cache_642626 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/adam/Library/Arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major -prefs=runtime.tools.arm-none-eabi-gcc-8-2018-q4-major.path=/Users/adam/Library/Arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major -verbose /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries/RTC/examples/Example1_Get_Time/Example1_Get_Time.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/adam/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/adam/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/adam/Documents/Arduino/libraries -fqbn=SparkFun:apollo3:sfe_artemis_mm_pb:svl_baud=921600,loader=option_svl -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/hm/mbygfq0s3g1f3vpv6dg01c7w0000gn/T/arduino_build_589717 -warnings=none -build-cache /var/folders/hm/mbygfq0s3g1f3vpv6dg01c7w0000gn/T/arduino_cache_642626 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/adam/Library/Arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major -prefs=runtime.tools.arm-none-eabi-gcc-8-2018-q4-major.path=/Users/adam/Library/Arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major -verbose /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries/RTC/examples/Example1_Get_Time/Example1_Get_Time.ino
Using board 'sfe_artemis_mm_pb' from platform in folder: /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5
Using core 'arduino' from platform in folder: /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5
Detecting libraries used...
/Users/adam/Library/Arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/arm-none-eabi-g++ -include /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/variants/SFE_ARTEMIS_MM_PB/mbed/mbed_config.h -include /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/arduino/sdk/ArduinoSDK.h -iprefix /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/ @/Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/variants/SFE_ARTEMIS_MM_PB/mbed/.cxx-flags -w -x c++ -E -CC -DARDUINO=10813 -DARDUINO_APOLLO3_SFE_ARTEMIS_MM_PB -DARDUINO_ARCH_MBED -DARDUINO_ARCH_APOLLO3 -DMBED_NO_GLOBAL_USING_DIRECTIVE -DCORDIO_ZERO_COPY_HCI -I/Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/arduino -I/Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/variants/SFE_ARTEMIS_MM_PB -I/Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/arduino -I/Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/arduino/mbed-bridge -I/Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/arduino/mbed-bridge/core-api @/Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/variants/SFE_ARTEMIS_MM_PB/mbed/.includes /var/folders/hm/mbygfq0s3g1f3vpv6dg01c7w0000gn/T/arduino_build_589717/sketch/Example1_Get_Time.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for am_bsp.h: []
ResolveLibrary(am_bsp.h)
  -> candidates: []
In file included from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/mbed-os/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/device/device.h:38,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/mbed-os/hal/ticker_api.h:25,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/mbed-os/features/netsocket/SocketStats.h:24,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/mbed-os/features/netsocket/InternetSocket.h:30,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/mbed-os/features/netsocket/UDPSocket.h:23,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/mbed-os/features/netsocket/nsapi.h:40,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/mbed-os/mbed.h:26,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/arduino/mbed-bridge/Arduino.h:11,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/arduino/sdk/ArduinoSDK.h:9,
                 from <command-line>:
/Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.5/cores/mbed-os/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/device/objects.h:20:10: fatal error: am_bsp.h: No such file or directory
 #include "am_bsp.h"
          ^~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Artemis MicroMod Processor.
```

Cheers,
Adam

adamgarbo, 2021-02-26:
Thanks @kirk-sfe for labelling this a bug. I should clarify that the board definitions are missing from the Mbed OS core:

`/mbed-os-ambiq-apollo3/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/`

kirk-sfe, 2021-02-26:
Yes - this got missed by the team.  We'll be focusing on the v2 core for the next 2-3 weeks to resolve issues like this on the core (as well as the macOS and Linux problems).  Look for a v2.1 release sometime in March.

adamgarbo, 2021-02-26:
Great news! Looking forward to it. I'm shifting my efforts to migrating my projects to v2.x, and am happy to help where I can.

Wenn0101, 2021-03-16:
Scheduled for v2.0.6, see https://github.com/sparkfun/Arduino_Apollo3/discussions/368 to preview release.

closed 2021-03-22 by Wenn0101