sparkfun/Arduino_Apollo3 #346: Missing WString.h

jerabaul29, 2021-02-09:
I get quite many messages about missing WString.h. Suppose it is one of these 'common' headers that should be part of the core? :) .

paulvha, 2021-02-10:
WString.h is included.....2.0.5/cores/arduino/mbed-bridge/core-api/api/deprecated BUT this folder is not included in search path. That said: looking in WString.h:   including WString.h is deprecated, for all future projects use Arduino.h instead
So apparently you can just delete an  '#include <Wstring.h>' in the source  as Arduino.h is included by default.
Regards,
Paul

jerabaul29, 2021-02-11:
Ok, thanks! This is slightly inconvenient as it breaks some libraries, but will be able to fix it then. Thanks! :)

closed 2021-02-11 by jerabaul29