sparkfun/Arduino_Apollo3 #347: using Wire1 on the core v 2.0.5 [probably needed to program the Artemis Global Tracker]

jerabaul29, 2021-02-11:
I think (to be confirmed there by the experts :) ) that I need to be able to access specifically Wire1 to be able to use the GPS on the Artemis Global Tracker (see issue https://github.com/sparkfunX/Artemis_Global_Tracker/issues/18 .

In the core v 2.0.5, doing:

```cpp
#include <Wire.h>

void setup(){
  Wire1.begin();
}
```

as was doable in the core v2 seems to be impossible:

```cpp
exit status 1
'Wire1' was not declared in this scope
```

- any way to get this to work already in core v 2.0.5?
- if not, can this be fixed? Any timeline? I really need this fix badly, any "hack" that could make it work?

jerabaul29, 2021-02-11:
Ok, looking here:

https://github.com/sparkfun/Arduino_Apollo3/blob/fbb2eaca27245627bf22b59f43faa232635bcd01/libraries/Wire/src/Wire.h#L68-L85

and there:

https://github.com/sparkfun/Arduino_Apollo3/blob/fbb2eaca27245627bf22b59f43faa232635bcd01/libraries/Wire/src/Wire.cpp#L161-L178

gives me good hope that this should be an easy fix.

jerabaul29, 2021-02-11:
Ok, got some help on the tracker website, it should be possible to set up a I2C with custom pins, looking at it now.

jerabaul29, 2021-02-12:
Ok, more problems coming up due to wrong pinmaps; taking the discussion on the Global Tracker issue though, feel free to help there.

jry-slc, 2021-03-03:
This is probably another symptom of the same bug.
Echoing my post to on the sparkfun forum: Home . SparkFun . Artemis . Artemis Arduino Core
Using Wire1 on Thing Plus
---
Editing variant.h will get the second I2C device created, and the Arduino IDE compiles and links. But now there's a runtime mbed-OS error
++ MbedOS Error Info ++
Error Status: 0x80010130 Code: 304 Module: 1
Error Message: pinmap not found for peripheral
Location: 0x26903
Error Value: 0x10
Current Thread: main Id: 0x10004C70 Entry: 0x28031 StackSize: 0x1000 StackMem: 0x100068F0 SP: 0x1000782C
For more info, visit: https://mbed.com/s/error?error=0x800101 ... THING_PLUS
-- MbedOS Error Info --

#if 1 // 2021-02-26 jry
#define VARIANT_WIRE_INTFCS 2
// Wire1 Pins
#define VARIANT_Wire1_SCL 17
#define VARIANT_Wire1_SDA 16
// end 2021-02-26 jry
#else // original
#define VARIANT_WIRE_INTFCS 1
#endif

Wenn0101, 2021-05-12:
Alright. I went back thru the targets and added wire interfaces for some of the Wire interfaces that were labelled on the board but not defined in the software. This means that @jry-slc Wire1 should be available on the thing plus without you needing to define it yourself.

I wanted to clarify how to define a custom Wire we don't have one defined on the pins you would like already.
TwoWire myWire(mySDA, mySCL); 
//Name this anything you would like, myWire chosen in this example
//No need to change VARIANT_WIRE_INTFCS this is used internally when defining built in wire
//No need to use MbedI2C type, TwoWire will work
//Make sure to begin before using
void setup() {
  myWire.begin();
}

closed 2021-05-12 by Wenn0101