sparkfun/Arduino_Apollo3 #349: using custom UART config [needed to program the Artemis Global Tracker]

jerabaul29, 2021-02-12:
I have a hard time getting the Iridium on the Artemis Global Tracker to work with the core 2.0.5. See https://github.com/sparkfunX/Artemis_Global_Tracker/issues/19 .

After some debugging, seems this comes from the need to set up a custom serial. This code does not produce a serial output on my computer:

```cpp
// with this line, do not even get the Serial output
// this is a custom serial for talking to the iridium modem, similar to AGT example 14
UART iridiumSerial(1, 25, 24);

void setup(){
  Serial.begin(115200);
  delay(10);
  Serial.println(F("booted"));
}

void loop(){
  
}
```

Any idea how to fix this?

jerabaul29, 2021-02-12:
Ok, even something like this fails:

```cpp
void setup(){
  Serial.begin(115200);
  delay(10);
  Serial.println(F("booted"));

  Serial1.begin(57600);
  delay(10);

  Serial.println(F("done with serial1"));
}

void loop(){
  
}
```

This prints the first message, but the second message never gets printed.

jerabaul29, 2021-02-12:
Even this does not work (second message not printed)

```cpp
void setup(){
  Serial.begin(115200);
  delay(10);
  Serial.println(F("booted"));
  Serial.end();

  Serial1.begin(57600);
  delay(10);
  Serial1.end();

  Serial.begin(115200);
  delay(10);
  Serial.println(F("done with serial1"));
}

void loop(){
  
}
```

jerabaul29, 2021-02-12:
I had a look at the core, and I am really confused about how the following things are implemented:

- the fact that there are several hardware serials (2 on this board)

vs

- the fact that each hardware serial can be routed to different pins (RX, TX, and flow control).

Could you extend the serial example to show how a specific hardware serial (1 and 2) can be routed to a specific set of (TX, RX) pins?

paulvha, 2021-03-01:
Sharing the experience from the good time I spend to understand what is happening, but have not found the solution yet. The problem starts as soon as Serial1 (or any next UART()) is beginning. At the end of each begin(), in hardwareSerial.cpp, it will call attach() for &UART::rxISR and only for RxIrq (meaning receive). This callback is set with attach() in Mbed-drivers Serialbase.cpp. After that Serial will NOT provide any output anymore.  BUT I do not understand why... yet. The logic in the program flow is complex, but looks OK to me. I have also changed the way hardwareSerial.cpp is working, applying what Arduino Nano 33 BLE does, but still, no change. It somehow looks to me that the attach() call for a second Serial is overwriting the rxISR call back for (the first) Serial or it is overwriting other memory. Letting a second serial perform all init-actions except an attach() call, the first Serial will continue to work (of course you can NOT receive any data on the second serial). Maybe this helps..
regards,
Paul

paulvha, 2021-03-05:
This was darn hard to debug and took a long time..but in the end an easy way to reproduce/overcome. 

If RX1 is "open", nothing connected  it will constantly generate input/interrupts, causing an unresponsive system.

Connect a 10K resistor between  +3v3 and RX1... the whole problem is gone. Connect a device to RX1.. the problem is gone.

**Connect a 10K resistor between GND and RX1 or leave it floating... you are back in trouble.** You do not have this issue on RX0 as the USB interface is connected. 

The pin_config  for the **RX1 pin needs to get an internal pull-up**. Sharing the root causefor now and looking where it is the best place to set the pull-up.  
regards,
Paul

jerabaul29, 2021-03-06:
Many thanks for your hard work, looking forward to the fix :) .

rserranosmith, 2021-03-08:
I'm sorry, I'm super new to this, I was able to change the PeripheralPinConfigs.c file, but I am unsure of where and how to recompile the limbed-os.a library. Any help would be very much appreciated, thank you for this solution!

paulvha, 2021-03-08:
I had the same question some time ago and got instructions https://github.com/sparkfun/Arduino_Apollo3/issues/279.  If you are new to this.. it will be a challenge to get it going but you can try, else what board are you using maybe I can create one for your board and sent you a link.
regards,
Paul

rserranosmith, 2021-03-08:
Thank you Paul! That would be super helpful! I am using the Artemis Nano Board!

jerabaul29, 2021-03-30:
So is that solved / were you able to add the pullups settings to the binaries shipped with the package? :)

Wenn0101, 2021-05-12:
Fix included v2.1.0

closed 2021-05-12 by Wenn0101

bimalpaneru, 2024-02-14:
Following  up on this issue,   I have been trying to use Pin AUD_BCLK for Tx and AUD_LRCLK for Rx for a RS485 module.

Been having the issue during the receiving. Nothing is being received on the Rx line whereas TX seems perfectly fine. I do not even know what might be causing this, think this is the closest thread to my problem.

I have 10k pullup on the Rx line in saying that with or without pullup nth changes. 

I am using a micromod variant of artemis.

Would really appreciate any suggestions on potential solution to this.