sparkfun/Arduino_Apollo3 #350: I2C example does not compile

jerabaul29, 2021-02-12:
The I2C example (in Arduino IDE, File > Example > Apollo3 > I2C) does not compile for me using the ```RedBoard Artemis``` board:

```
I2C:65:15: error: 'D25' was not declared in this scope
 #define mySDA D25
               ^~~
/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries/Apollo3/examples/I2C/I2C.ino:67:16: note: in expansion of macro 'mySDA'
 MbedI2C myWire(mySDA, mySCL);
                ^~~~~
/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries/Apollo3/examples/I2C/I2C.ino:65:15: note: suggested alternative: 'D21'
 #define mySDA D25
               ^~~
/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries/Apollo3/examples/I2C/I2C.ino:67:16: note: in expansion of macro 'mySDA'
 MbedI2C myWire(mySDA, mySCL);
                ^~~~~
I2C:66:15: error: 'D27' was not declared in this scope
 #define mySCL D27
               ^~~
/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries/Apollo3/examples/I2C/I2C.ino:67:23: note: in expansion of macro 'mySCL'
 MbedI2C myWire(mySDA, mySCL);
                       ^~~~~
/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries/Apollo3/examples/I2C/I2C.ino:66:15: note: suggested alternative: 'D21'
 #define mySCL D27
               ^~~
/home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries/Apollo3/examples/I2C/I2C.ino:67:23: note: in expansion of macro 'mySCL'
 MbedI2C myWire(mySDA, mySCL);
                       ^~~~~
Using library Wire at version 2.0.0 in folder: /home/jrlab/.arduino15/packages/SparkFun/hardware/apollo3/2.0.5/libraries/Wire 
exit status 1
'D25' was not declared in this scope

```

mjaffee1997, 2021-02-14:
Try SDA on pin 14 and SCL on pin 15 (looking at the schematic). In any case, I haven't had any luck using the I2C pins on-board. Breaking out the QWIIC SDA and SCL pins works ok. These are configured differently in the core so one might be working while the other is not?

Wenn0101, 2021-03-16:
Scheduled for v2.0.6, see https://github.com/sparkfun/Arduino_Apollo3/discussions/368 to preview release.

closed 2021-03-22 by Wenn0101