sparkfun/Arduino_Apollo3 #351: Where to get I2C buffer size?

nseidle, 2021-02-12:
v2 core breaks this [external EEPROM library](https://github.com/sparkfun/SparkFun_External_EEPROM_Arduino_Library/blob/master/src/SparkFun_External_EEPROM.h#L34) because the define name changed away from the HAL to mbed.

We have libraries that need to check the size of the I2C buffer of a given platform in order to optimize comm. Assuming 32 bytes can be inefficient to ineffective depending on the device (u-blox modules, SPS30, I2C EEPROM, etc should have very different behavior depending on how big the buffer is).

It looks like the [`usedTXBuffer`](https://github.com/sparkfun/Arduino_Apollo3/blob/master/libraries/Wire/src/Wire.cpp#L8) is hardcoded to 256. If at all possible, make this variable and user settable. If not possible, then what is the define to check against to determine the buffer size?

paulvha, 2021-02-13:
I run into the same issue on  MKRZERO,  SAMD and  Apollo3. In Wire.h (r/libraries/Wire/src/Wire.h) it is set as
    RingBufferN<256> rxBuffer;
    uint8_t txBuffer[256]; 

The only way around was to determine / hard code the settings based on the platform as you have done in EEPROM library.
regards,
Paul

Wenn0101, 2021-05-12:
Added a macros AP3_WIRE_RX_BUFFER_LEN and AP3_WIRE_TX_BUFFER_LEN in https://github.com/sparkfun/Arduino_Apollo3/commit/9c1bcac460fe326c23badd16fa9228833f22b794

closed 2021-05-12 by Wenn0101