sparkfun/Arduino_Apollo3 #354: I2C timeout

jerabaul29, 2021-02-18:
Is there a way to set an I2C timeout, similar to what is discussed in:

https://github.com/arduino/Arduino/issues/10803

?

I looked for it in the documentation, but did not find anything about it :) .

Wenn0101, 2021-03-02:
I will have to look into this functionality

paulvha, 2021-12-16:
Recently I completed a deep-dive on I2C on Apollo3 and looked at this. A timeout is already in place, but you can not adjust it. There is no reference call in MBED-OS that allows you to do that.

For Arduino AVR the complete handling of the I2C protocol is done asynchronously in twi.c. The main processor is handling the sending of the start, stop, handle clock stretching, ACK & NAK signals as well as sending each byte by working directly with the different registers of the CPU. It is also checking that all steps are handled correctly. The chance of something going wrong and thus blocking the main processor is indeed a risk for which setting timeout makes sense. Looking at earlier versions on the Arduino AVR (before 1.8.2), there was NO timeout check at all.

In an Apollo3 all I2C communication is handled by Input-Output-Module (IOM4 for Wire). It is working independently of the main processor. It is much more robust as it is focused on one task only and it was built for that.  Wire() is using blocking transfer, which means the library is waiting for the IOM to complete the transfer by checking the status register of the IOM. There is already a timeout included of ~0.5s. If sending or receiving is not completed within that time, it will return an error. There is also a non-blocking transfer option, but that is not used by Wire().

closed 2022-03-17 by Wenn0101