sparkfun/Arduino_Apollo3 #358: any more information about the burst mode?

jerabaul29, 2021-02-21:
I have been looking at both the datasheet and the Sparkfun documentations about the burst mode on the Ambiq Apollo 3 / Artemis. I find it hard to get any information (except that it brings frequency to 96MHz and uses a bit more power).

Has anybody:

- a better source of information?
- any idea why it is called "burst" mode? Should it be used for no more than a few micro seconds at a time? milliseconds at a time? seconds at a time? Can it be used for as long as wanted?
- any idea why this is a burst mode and not just a "dynamic clocking"? Are there some implications there?
- anything more a user should know about the burst mode?

klsorensen, 2021-02-25:
In version 1.2.1 of the Apollo3 Boards there was an:

enableBurstMode(); 

and 

disableBurstMode();

function that you could call. After version 1.2.1, these functions were not available. . .  not sure why.

Marsfan, 2021-03-04:
@jerabaul29 This is anecdotal, but I have had code that stays in burst mode for the entirety of `void loop()`, and it seemed to run fine.

Marsfan, 2021-03-04:
@Wenn0101 @klsorensen  I figured I would mention this here since #318 was closed. I wrote the following Arduino program to try to use the burst mode functions in AmbiqSuite SDK:

<details>
<summary> turboTest.ino </summary>

```Arduino
am_hal_burst_avail_e burstAvail;
am_hal_burst_mode_e burstStatus;

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  am_hal_burst_mode_initialize(&burstAvail);
  Serial.begin(115200);
}

// the loop function runs over and over again forever
void loop() {
  am_hal_burst_mode_enable(&burstStatus);
  Serial.print("Hello");
  am_hal_burst_mode_disable(&burstStatus);
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```
</details>

Running this program triggered an Hard Fault on my Artemis ATP, and MBED OS printed out the following error dump:

<details> 
<summary> MBED OS Error Dump </summary>

```
++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0: 40004034
R1: 0
R2: 1
R3: 10000F09
R4: 10000FBD
R5: 10000FBD
R6: 0
R7: 0
R8: 0
R9: 0
R10: 0
R11: 0
R12: 8
SP   : 10006BB0
LR   : 25E11
PC   : 10000F08
xPSR : 61000000
PSP  : 10006B90
MSP  : 1005FF70
CPUID: 410FC241
HFSR : 40000000
MMFSR: 1
BFSR : 0
UFSR : 0
DFSR : 0
AFSR : 0
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x10000F08
Error Value: 0x10005B80
Current Thread: main Id: 0x10004120 Entry: 0x2273D StackSize: 0x1000 StackMem: 0x10005BE0 SP: 0x10006BB0 
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=SFE_ARTEMIS_ATP
-- MbedOS Error Info --
```

</details>

I managed to do some digging around with a SWD Debugger for step through testing, and it appears that the error occurs when `  am_hal_burst_mode_disable(&burstStatus);` is executed. Based on what I am seeing in the debug registers, it appears to be an _Instruction Access Violation_ that is triggered during execution of `am_hal_flash_store_ui32`. 

I'm really not sure whether this is a mistake in my program, or an issue in the AmbiqSuite SDK, or an issue with MBED OS. I am leaning towards an MBED OS related issue, since the error output is from MBED OS, and I used pretty much the same syntax  that is in the v1 branch. 

What is really weird is that the Artemis seems to be still doing stuff after the hard fault. The LED blinks a few times fast and a few times slow. I vaguely remember writing a program that did that for my board, so I wonder if something in the flash is acting weird.

Wenn0101, 2021-05-12:
Initial Burst Mode implementation included in v2.1.0

closed 2021-05-12 by Wenn0101

Marsfan, 2021-05-12:
@Wenn0101 I think this should stay open as the questions asked by the OP are still valid.