sparkfun/Arduino_Apollo3 #360: max baudrate

jerabaul29, 2021-02-22:
I want to try communicating with the BNO080 through serial. The datasheet there mentions that the BNO080 baudrate on serial is 3Mb/s, Any idea what baud rate that would correspond to, and if the Artemis is able to attain it?

idea--list, 2021-02-23:
@jerabaul29 
To find out the max baudrate the board can handle use this code:

```
void setup(){};

void loop()
{

  delay(1000);
  Serial.begin(57600);
  Serial.println("\r\rBaud-rate = 57600");
  delay(1000);
  Serial.begin(76800);
  Serial.println("\r\rBaud-rate = 76800");
  delay(1000);
  Serial.begin(115200);
  Serial.println("\r\rBaud-rate = 115200");
  delay(1000);
  Serial.begin(230400);
  Serial.println("\r\rBaud-rate = 230400");
  delay(1000);
  Serial.begin(250000);
  Serial.println("\r\rBaud-rate = 250000");
  delay(1000);
  Serial.begin(500000);
  Serial.println("\r\rBaud-rate = 500000");
  delay(1000);
  Serial.begin(1000000);
  Serial.println("\r\rBaud-rate = 1000000");
  delay(1000);
  Serial.begin(2000000);
  Serial.println("\r\rBaud-rate = 2000000");
};
```

jerabaul29, 2021-02-24:
Ok, thanks, seems that it hangs from 2M baudrate and over.

closed 2021-02-24 by jerabaul29