sparkfun/Arduino_Apollo3 #365: How is integration on the Arduino IDE 2.0

jerabaul29, 2021-03-06:
It looks like the Arduino IDE 2.0 will bring many improvements that will make life much easier for developers: autocomplete, integrated debugging, etc.

They name that it should compatible with SAMD and Mbed boards:

https://blog.arduino.cc/2021/03/01/announcing-the-arduino-ide-2-0-beta/

Does that mean it should work with this core? That would've awesome.

I will try to find the time to test some day, but if someone already has tried, would be very interested to hear :) .

idea--list, 2021-03-09:
Yes Arduino IDE V2.0 Beta3 works with Artemis Core V2.0.5. Until now i only tried the blinky example which compiles&runs fine.
The new IDE introduces linting and it is possible to create .h/.cpp files beyond the the default .ino file.
The beta however does not still have the plotter and seems to be somewhat slow here and there. Sometimes i have to unplug and replug the board to get it recognized.

However it made me considering to convert from Mbed OS to Arduino and that is huge! I mean i could not imagine that before.

jerabaul29, 2021-03-11:
I can confirm (tested just now) that things work well between the cores in both v1 and v2 and the Arduino IDE 2.0. This is amazing, finally an IDE that is usable out of the box :) .

jerabaul29, 2021-03-11:
PS: @Wenn0101 you may want to consider updating tutorials some day and suggesting using the Arduino IDE 2.0 instead of Arduino IDE 1.8. It is such a huge difference for user experience, really day and night. It was very easy to install (see the binaries linked from the readme, section ```Nightly builds``` here: https://github.com/arduino/arduino-ide ).

closed 2021-03-11 by jerabaul29

amirna2, 2021-07-15:
I am glad to hear the UX is better. I am eager to try Arduino IDE 2.0 as dealing with the 1.8 version is so frustrating.
I am also noticing that compilation on the Apollo3 platforms is extremely slow compared to other platforms. 
Is this something anyone has noticed as well? Is it better with Arduino 2.0?
It takes about 11 minutes to compile my program the first time and about 5 minutes the second time (after caching).
But on other platforms I have tried it usually is less than a minute.