sparkfun/Arduino_Apollo3 #369: Highest Baud rate setting for SVL on macOS and linux fail

kirk-sfe, 2021-03-16:
Uploading to an Artemis board on macOS or Linux using the SVL fails when the baud rate is set to 921600. Firmware upload works if the baud rate is lowered to 460800. 

**Error Message**
Target failed to enter bootload mode. Verify the right COM port is selected and that your board has the SVL bootloader.

This should be noted / tied to an issue on the SVL boot loader. 

ALSO: The error message should be fixed and not reference a "COM" port, which are non-existent on macOS/Linux.

adamgarbo, 2021-03-24:
I would also add that if the 921600 baud rate doesn't end up being supported, it would be helpful for users to either simply remove it, or change the order of the baud rates so that the Arduino IDE doesn't default automatically to 921600 when changing the selected board. This is a minor annoyance I deal with on a daily basis.

Wenn0101, 2021-03-24:
@adamgarbo If this is not fixed in the next release, we will definitely have the default changed. I agree that this is an annoyance.

paulvha, 2021-03-29:
just to add 921600 works consistent without problems on my Edge board with an FTDI Friend from Adafruit. The issue is with the CH340 chip/driver which can not handle this speed.  I have tried before CH340 board before switching and that fails constantly. 

regards,
Paul

Wenn0101, 2022-03-17:
For now the workaround seems to be sufficient. Closing this issue.

closed 2022-03-17 by Wenn0101

adamgarbo, 2022-03-17:
@Wenn0101 which workaround are you referring to? Having the 921600 baud rate kicked to the end of the list is a nice change (although I still use v1.2.3 where it's still at the top of the list).