sparkfun/Arduino_Apollo3 #372: Serial1 Doesn't compile for the Artemis Nano board - v2.0.6

kirk-sfe, 2021-03-23:
Serial1 doesn't compile for the Artemis Nano w/ v2.0.6

Looks like a config issue in the Nano variant/board files. If you dig into the Apollo3 implementation,  ../Arduino15/packages/SparkFun/hardware/apollo3/2.0.6/variants/SFE_ARTEMIS_NANO/variant.cpp and uncomment the Serial1() function declaration, it will compile. Not sure if it will work though ...

Serial1 for all Artemis variants needs to be checked and validate.

paulvha, 2021-04-04:
FYI. there have been a number of posts (e.g. https://forum.sparkfun.com/viewtopic.php?f=169&t=54315&p=220736&hilit=Nano+Serial1#p220736) where Serial is NOT working when Serial1 is enabled. The only solution was, comment out Serial1 definition if there is NOTHING connected to the Serial1 port. 
The problem can be reproduced and the reason + solution is documented in https://github.com/sparkfun/Arduino_Apollo3/issues/349.
regards,
Paul

kirk-sfe, 2021-04-05:
Thanks. The serial/serial1 issues are being worked on for the next release - 2.1. Hopefully out by end of month.

closed 2021-05-12 by Wenn0101