sparkfun/Arduino_Apollo3 #373: SVL permissions error present on v2.0.6

adamgarbo, 2021-03-23:
Hi folks,

After updating to v2.0.6, I experienced the same svl permissions error on both macOS Big Sur and Catalina.

However, when I tested the preview of the release by manually downloading the core and renaming it to 2.0.4, this issue didn't present itself.

After a `chmod +x svl` I can upload code successfully, but it's strange that the fix didn't seem to stick for v2.0.6. 

Would be helpful if any other macOS or Linux users could confirm this (e.g. @jerabaul29).

Cheers,
Adam

Wenn0101, 2021-03-24:
@adamgarbo Give it another go.

I repackaged the release, and I have it working on my test machine.

adamgarbo, 2021-03-27:
Removed and redownloaded v2.0.6 and was able to upload code on macOS Big Sur without any svl permission errors. Thanks, @Wenn0101!

closed 2021-03-27 by adamgarbo