sparkfun/Arduino_Apollo3 #375: Repackage v1.2.1 with updated artemis_svl

adamgarbo, 2021-03-27:
Hi @kirk-sfe,

I was able to follow your [instructions](https://github.com/sparkfun/Arduino_Apollo3/issues/299#issuecomment-779335387) to get v1.2.1 working on macOS Big Sur by generating a new artemis_svl file. Thanks for the help.

Would it be possible for SparkFun to have v1.2.1 repackaged with an updated artemis_svl file? Until v2.x is more stable, many users (and SparkFun projects) actually depend on v1.x of the Core.

I also just submitted a PR (https://github.com/sparkfun/Arduino_Apollo3/pull/374) to v1 with a couple of minor updates, which could be bundled up with the new artemis_svl.

Cheers,
Adam

Wenn0101, 2021-03-31:
Thanks for the PR, and suggestion. It may be time for a v1.2.2, I will take a look at this right after finishing up v2.1.0.

Wenn0101, 2021-07-20:
Cleaning up some issues I never closed. I believe this was done in 1.2.2.

closed 2021-07-20 by Wenn0101