sparkfun/Arduino_Apollo3 #377: Upgrading between version using arduino IDE fails on first attempt

Wenn0101, 2021-03-31:
When a previous version of the core is installed, upgrading to the latest may fail the first attempt. The first download shows a bogus size (600MB+, but starts at 300MB+) and fails the CRC check. 

Retrying immediately will result in a successful install on the next attempt.

Still yet to determine if this is a problem with arduino, our json file, or our file storage.

paulvha, 2021-04-07:
HI,

The issue is related that starting 2.x version the **same tar-name is used for each version: Arduino_Apollo3.tar.gz.** Opposite to V1.x where the filename contained a version e.g. v1.2.1.tar.gz. 

I installed 2.0.4. the tar size is 325,830, then I updated to the latest 2.0.6 and see the bogus size of 683,492. This 683,492 is the size of the 2.0.4 tar (325,830) + the size of 2.0.6 (357,662). As it fails, the size of Arduino_Apollo3.tar.gz shows 683K. So the downloaded 2.0.6 is appended to the original 2.0.4 tar-file. After it fails and trying for a second time it downloads the right size and installs. 

I then installed 2.0.3 with tar-size 325,9 and update to 2.0.4 which has tar-size 325,83 (smaller!!). Works without problems. Now update to 2.0.5 which has tar-size 348,333(larger). It then failed the first time as it tried with a bogus size of 348,333 + 325,833.

I expect this to be an issue in the ArduinoIDE. In FileDownloader.java, there is the function downloadFile(boolean noResume). I suspect that if the tar-file is existing BUT smaller it will try to resume the download and append. I have not tried to debug that further as it would take me too much time. 

A fast workaround/solution is to **use a different tar-filename for each version**. Looking at the other files in staging I see they all have a version number in them.. but that said, they are all much smaller.

Using the same name helps to save package-staging disc-space.  The disadvantage is when switching between 2.x versions the tar-file needs to be downloaded each time.  The size of the Sparkfun package is around 350K. Next to taking a large amount of disk-space, it takes longer to download and long to install. 

**How to reduce?**
The 2 BIG drivers are Variants and Cores
**_Variants:_**
Mainly driven by Mbed-os library. 
Use the Q42020 compiler (we will have to for the virtual issue with ArduinoBLE). A single Mbed-os library for each board is 47MB instead of 60MB (13MB * 9 board = 108MB less or about 15MB on the TAR file) 

**_Cores_**
Mainly driven by Mbed-Os source code. 
Reduce the inclusion of the different mbed-os/targets that we do not use. E.g. the targets folder (uncompressed) is 500MB, of which TARGET-STM is already 250MB (AND NOT USED AT ALL). We actaully on need TARGET_Ambiq_Micro, which is 11MB.. 

OR as others do, make the Mbed-os source files available to download separately. Keep them on github, but do not include the distribution tar-file. save another 30 to 40MB on the TAR file)

regards,
Paul

Wenn0101, 2021-07-20:
this seems to happen less with smaller package sizes. I am closing this for now due to reduced size. Future releases will likely be smaller.

closed 2021-07-20 by Wenn0101