sparkfun/Arduino_Apollo3 #378: Cannot upload from Ubuntu 16.04

miquelmassot, 2021-04-02:
Hi,

I have got an Artemis Global Tracker connected to a PC running Ubuntu 16.04, and I managed to compile a sketch for the board, but when I tried to upload it using Arduino IDE 1.8.13, I get the following error:

    Sketch uses 73472 bytes (7%) of program storage space. Maximum is 960000 bytes.
    Global variables use 58060 bytes of dynamic memory.
    An error occurred while uploading the sketch
    [11746] Error loading Python lib '/tmp/_MEIecOVmZ/libpython3.6m.so.1.0': dlopen: /lib/x86_64-linux-gnu/libc.so.6: version 
    `GLIBC_2.25' not found (required by /tmp/_MEIecOVmZ/libpython3.6m.so.1.0)

is it possible that the pyinstaller of your application was compiled with glibc version 2.25 and I have got 2.23 and I cannot upload. Is it possible to have a 2.23 version to be compatible with ubuntu 16.04?

kirk-sfe, 2021-04-06:
This is fixed and should be in the next release (in a couple weeks). 
- Moved our SVL build environment to Ubuntu 16.04 (was at 20.04)

If you want to try it now:
- copy the svl at:  
     https://github.com/sparkfun/Apollo3_Uploader_SVL/blob/master/dist/linux/svl
- to this location:
    ~/.arduino15/packages/SparkFun/hardware/apollo3/2.0.6/tools/uploaders/svl/dist/linux

This worked on a fresh 16.04 install

miquelmassot, 2021-04-12:
Hi @kirk-sfe, thanks for the quick reply!
I am using the boards defined in "SparkFun Apollo3 Boards" v1.2.0 as this is the only version I got to work. I am using the code examples for the [Artemis Global Tracker](https://github.com/sparkfunX/Artemis_Global_Tracker) ([board](https://www.sparkfun.com/products/16469)) and the version you sent me is for v2.0.6 which uses a completely different system (mbedOS) and the code I have is no longer compatible. Would it be possible to get v1.2.0 for an ubuntu 16.04 version?

kirk-sfe, 2021-04-12:
Hi - 
It shouldn't be an issue. In one of our upcoming sprints, we updating SVL in 1.2 for Mac, so we can add Linux. I just added that story to our backlog.

closed 2021-07-20 by Wenn0101