sparkfun/Arduino_Apollo3 #381: question about the version of the underlying MbedOS core

jerabaul29, 2021-04-08:
This may be related to some previous questions, but I was curious of the following.

It looks like the underlying version of MbedOS core is quite old: https://github.com/sparkfun/mbed-os-ambiq-apollo3/tree/0773105959af71b6d21a6f722635d837b49ba344 

If I am not confused, the latest updates coming from upstream MbedOS are from mid-2020: https://github.com/sparkfun/mbed-os-ambiq-apollo3/commits/0773105959af71b6d21a6f722635d837b49ba344?after=0773105959af71b6d21a6f722635d837b49ba344+69&branch=0773105959af71b6d21a6f722635d837b49ba344

Any reason for not upgrading the underlying MbedOS core? I suppose there have been quite many improvements and fixes in the last year?

jerabaul29, 2021-04-08:
Would be exciting, would be great to benefit from MbedOS buxfixes and development.

paulvha, 2021-04-09:
I think Serial communication is getting a bigger issue. 

I was reading a great article about a big change in Mbed6 in the whole printf():  https://forums.mbed.com/t/hitchhikers-guide-to-printf-in-mbed-6/12492. Now it looks easy to just make the change in mbed_app.json.. however, this configuration file is NOT used when building a library, only when you compile a complete application. Also see #278

Next to that, it seems there is a difference in the way serial communication is handled between standard output and device serial. All explained in that article. I wonder about that impact. 

regards,
Paul

Wenn0101, 2021-07-20:
Closing this issue, as the question was answered. I will keep you all posted when large updates to the mbed core occur.

closed 2021-07-20 by Wenn0101