sparkfun/Arduino_Apollo3 #382: Redboard Artemis Thing Plus delay() twice as fast

halveez, 2021-04-10:
Had a previously programmed ATP board from August 2020 functioning as intended.

Recently had to reinstall all Arduino libraries and board manager files to computer. 

Loading the same program to the ATP board is running twice as fast (confirmed with a stopwatch demo, taking 15s to "count" to 30s)

ATP board also finnicky when first plugged in - turning off and back on within the first 1-3 seconds, seems very unstable. 

Looking into what may have changed with the Apollo3 version since August 2020 to cause the timing and connectivity issues.

halveez, 2021-04-10:
Board is disconnecting at a consistent time

halveez, 2021-04-10:
Modified stopwatch demo in use:

[stopwatch-test.zip](https://github.com/sparkfun/Arduino_Apollo3/files/6290274/stopwatch-test.zip)

halveez, 2021-04-10:
Modified demo for a stopwatch function here.
[stopwatch-test.zip](https://github.com/sparkfun/Arduino_Apollo3/files/6290277/stopwatch-test.zip)

halveez, 2021-04-10:
Using Apollo3 v1.2.1 and MicroOLED v1.2.0 takes 30 seconds to count to 20 on screen

halveez, 2021-04-10:
Using Apollo3 v2.0.6 and MicroOLED v1.2.0 does not work - need to install more recent version of MicroOLED library

halveez, 2021-04-10:
Using Apollo3 v2.0.6 and MicroOLED v1.3.2 takes 10 seconds to count to 20 on screen

halveez, 2021-04-10:
Testing the standard LED blink demo with both Apollo3 1.2.1 and 2.0.6 appears to be running at a correct speed. 

Error likely coming from the Micro_OLED library.

idea--list, 2021-04-12:
@halveez 
Arduino core for Apollo3 v2+ is based on a late version of MbedOS 5.x release. While Arduino core v1.x is a pure Arduino solution. So under the hood they differ significantly.

In Mbed OS things tend to change rapidly though, and even v2.x of Arduino core contains some bugs. So try to stick with the versions which worked as intended. I hate to advise this as also i prefer to update the software framework, but strange things tend to occur after doing so.

halveez, 2021-04-12:
Thanks Peter.

Fortunately for my program it was only used as an inconsequential delay
between printing instructions, and not a computation or something that
needed accurate time tracking, so I was able to make a functional solution
quickly.

Agreed, and I should have downloaded and archived the specific libraries
that I used for this project initially to avoid such an issue.

Wenn0101, 2021-04-12:
Thanks for the report. I'll add it on to our list of things to check into.

If you are looking for a version of the core, but you kind find it on arduino anymore, you can check out the [releases page](https://github.com/sparkfun/Arduino_Apollo3/releases). If you still need it, you may be able to find the older version of the core that you are looking for. You can checkout to these tags if you download the source using git. It sounds like you may have been using v1.2.0 or v.1.1.2.

As mentioned. There have been some pretty significant changes since then. The v2 core works with an mbedOS core. The initial releases of these v2 core were admittedly a bit rocky, but we have been working thru the issues reported by users with the help of the community, and feedback I have gotten on the last few releases has been more positive.

Wenn0101, 2022-03-17:
This issue seems to have gone stale, and I havent had recent reports of delay() being off. Closing.

closed 2022-03-17 by Wenn0101