sparkfun/Arduino_Apollo3 #383: Pin Drive Strength

jerabaul29, 2021-04-21:
From https://learn.sparkfun.com/tutorials/designing-with-the-sparkfun-artemis :

"
Pin Drive Strength

One of the more unique features of the Artemis is its ability to have selectable drive strength on all GPIO. 2, 4, 8, or 12mA can be selected as the max current on a given pin. Additionally pads 3 and 36 have selectable high side power switch transistors to provide ~1 Ω switches to VDDH. Pads 37 and 41 have a selectable low side power switch transistors to provide ~1Ω switches to VSS.
"

A few questions here:

- how to select drive strength on individual pins? I have a project on which I would like to power / switch on and off a small sensor (needing around 5 mA), I suppose this means it should be possible to power it directly from a GPIO pin, right? Any example of how this can be set in code? (both in core v1 and v2?)

- do I understand well that pads 3, 36, 37, 41, have some even extra switching possibility? Does that mean I can take several 100s of mA from each of these pads? Any example of code for how to use them (by curiosity, is it something included in the Ambiq MCU itself, or is it because you added some components somewhere? :) ).

jerabaul29, 2021-04-21:
PS: and I cannot find where these 4 pads 3, 36, 37, 41 are broken out; are these related to the pads 24 to 29 broken on the bottom of the board?

Wenn0101, 2021-04-27:
@jerabaul29 Good questions. Let me take a look at this feature and see where its back. I'll let you know what I find.

paulvha, 2021-04-30:
I took some time to look at this. I have documented the findings, instructions and code so you can use more of the GPIO capabilities from a sketch. https://github.com/paulvha/apollo3/tree/master/gpiocontrol.

Not sure board you have, but on my ATP the 4 pins are next to the processor.

regards,
Paul

jerabaul29, 2021-04-30:
Many thanks, had a look at it, really great work @paulvha  :) .

Do you think this could be made into a small "library" to integrate into the core v1 and v2, so that people can use these pings from a high level point of view?

@adamgarbo this should answer some of the discussions we had had on some other channels :) .

Many thanks again, this is amazing, as this means that small "non power hungry" sensors etc can now be switched on off without any additional components :) .

paulvha, 2021-05-01:
well, I have made a library to install in the Sparkfun boards directory (either 1.2.1 or V2.06). It has 4 examples (including powering a BME280). https://github.com/paulvha/apollo3/tree/master/APgpio
regards,
Paul

jerabaul29, 2021-05-01:
This is amazing, many thanks :) .

A small question: all your code refers to pads and not numbered pins, right? So I suppose users should go to the schematics to know which pins that corresponds to which pads? :) Or is there a way to programmatically translate between pads and pin numbers?

Do you think this could be pushed to the "main libraries folder"? https://github.com/sparkfun/Arduino_Apollo3/tree/master/libraries

paulvha, 2021-05-02:
you are a demanding customer :-)

A good question however and indeed people should look at schematics.. BUT... to help ... I have created an overview of the different Artemis Boards mapping the pin-descriptions printed on the board to the pads. Also shows whether a pad is used on the board itself and not exposed externally of not used at all.  It is now added in a spreadsheet in 2 different formats in the extras- folder of the APgpio library.

Whether this can be part of the Apollo3 master is a not up to me. That is a Sparkfun decision.

regards,
Paul

jerabaul29, 2021-05-02:
Haha sorry 😇😂 . It is just that, in my experience, polishing a bit the usability has a lot to say for how many who actually manage to use / enjoy libraries etc :) .

That sounds good :) @Wenn0101 do you think this could become part of the Apollo 3 master? Would be very useful :) .