sparkfun/Arduino_Apollo3 #386: working with CMSIS: missing arm_const_structs.h: No such file or directory

jerabaul29, 2021-05-03:
When working with CMSIS functions (for my part, I am especially interested in the FFTs), one often needs stuff from arm_const_structs.h (for example, for FFTs, this contains pre computed twiddle factors etc). As a side note, for an example of how to use these pre computed tables, see https://github.com/ARM-software/CMSIS/blob/master/CMSIS/DSP_Lib/Examples/arm_fft_bin_example/GCC/arm_fft_bin_example_f32.c .

However this does not work at the moment, getting error messages on core v1.2.1 (I cannot move to core v2.X for my use at the moment):

```
fatal error: arm_const_structs.h: No such file or directory
 #include "arm_const_structs.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
arm_const_structs.h: No such file or directory
```

I suppose this is because the ```Include``` folder of CMSIS (not sure where it is for the core, but on CMSIS github repo, this is this one I think: https://github.com/ARM-software/CMSIS/tree/master/CMSIS/Include ) is not included / visible to the compiler, right? Could this be fixed / would it be possible to get a patch for both the core v1.2.1 and the core v2?

This may be quite similar to the issues found in #326 , @Wenn0101 if you can help on this one too it would be really great :) .

Wenn0101, 2021-05-04:
Hmm, this is an interesting one. I may need a bit to understand where how these files are supposed to be included. 

I'll let you know what I find.

jerabaul29, 2021-05-05:
Sounds good :) Getting fixed for this together with #387 would be amazing :) .