sparkfun/Arduino_Apollo3 #387: is an old version of CMSIS and DSP in use / any way to update it?

jerabaul29, 2021-05-03:
After discussion on the CMSIS repo, it seems that there may be some outdated version of the CMSIS DSP being shipped; see https://github.com/ARM-software/CMSIS_5/issues/1189 . Any way to update that? There are some issues with the version provided (for example, real FFTs failing for a FFT length 128).

Wenn0101, 2021-05-04:
One of the next items I will be working on is updating the mbed version we are using on the core to a more recent version. While I am updating, I'll make sure to check in on this and see if it is all up to date.

idea--list, 2021-05-04:
@Wenn0101 
While doing so, just avoid the 6.7 release as with that particular one [nothing works](https://github.com/ARMmbed/mbed-os/issues/14273) on the Artemis boards.