sparkfun/Arduino_Apollo3 #389: how to install from the board manager the v1 of the core?

jerabaul29, 2021-05-08:
It looks like only the core v2 is available on the Arduino cores, for example for the Arduino IDE2.x I get to install the core v2.0.3 to v2.0.6, but no trace of the core v1.x.x.

[image]

The same is true of the Arduino IDE 1.8.13, though for a reason I ignore, I cannot take a screenshot.

Is there something I am doing wrong, or has the core v1 disappeared from the Arduino board manager?

I have some code only running on the core v1, and the core v2 is still a bit too "experimental" for some of my applications at the moment, so I (and the people I work with) need the core v1. My question / request is then: could you make sure that the latest version of the core v1 is also published and available for install?

Wenn0101, 2021-05-08:
Hmm interesting. I still see it in Arduino 1.8.13, I wonder what the difference is in our set-up. what OS are you using?

I'll try it in the Arduino IDE 2.x

[image]

jerabaul29, 2021-05-08:
Many thanks for checking that out.

Ubuntu 20.04.

Aaah, sorry, maybe I understand: looks like I am on 1.2.1 already and that is the one I cannot see in the menu, while you are on 2.0.6 and that is the one you cannot see on the menu.

jerabaul29, 2021-05-08:
Sorry, that may be a false alarm then. Strange, I was checking it because someone told me they could not install the 1.xx. They were probably doing something wrong another place...

jerabaul29, 2021-05-08:
If you can confirm this then feel free to close. Sorry for the noise.

closed 2021-05-11 by Wenn0101