sparkfun/Arduino_Apollo3 #399: No Example of Serial Interrupts

YFNZ, 2021-05-17:
It'd be awesome if someone could add an example using serial interrupts to receive serial data onto the device.

Wenn0101, 2021-05-18:
Thanks for the feedback, we will take a look.

Wenn0101, 2021-06-10:
Hello, I am taking a look back at this issue to work on it. Can you explain more what you were hoping for in the example. 

The [existing serial example](https://github.com/sparkfun/Arduino_Apollo3/blob/main/libraries/Apollo3/examples/Serial/Serial.ino) uses interrupts but it all happens "under the hood". Were you hoping to be able to be able to perform logic in the interrupt yourself? were you hoping for an example that uses Serial1?

Let me know. Thanks.
-Kyle

paulvha, 2021-06-30:
looks to me this was related to this post. https://forum.sparkfun.com/viewtopic.php?f=169&t=55742
I think the issue is solved.
regards,
Paul

closed 2021-07-20 by Wenn0101