sparkfun/Arduino_Apollo3 #400: I2C sub-one cycle clock stretching causes problems. Me, I blame SMPCNT.

PaulZC, 2021-06-01:
The Qwiic Button uses sub-one cycle clock stretching on the I2C bus. This causes the standard ```beginTransmission``` ```endTransmission``` 'scanner' test for an I2C device to return a false positive.

**Steps To Replicate:**

Connect an unmodified Qwiic Button to e.g. an Artemis Thing Plus.
Load core v2.1.0.
Upload the following simple 'I2C scanner' sketch:

```
#include <Wire.h>

void setup()
{
  Serial.begin(115200);
  Wire.begin();
  Wire.setClock(100000);
  
  for (byte address = 1; address < 127; address++ )
  {
    Wire.beginTransmission(address);
    byte error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      Serial.println(address, HEX);
    }
    else
    {
      Serial.print("I2C error ");
      Serial.print(error);
      Serial.print(" at address 0x");
      Serial.println(address, HEX);
    }
  }
}

void loop()
{
}
```

The Qwiic Button has a default address of 0x6F, so you should see this (as captured with an ESP32 Thing Plus):

```
...
I2C error 2 at address 0x66
I2C error 2 at address 0x67
I2C error 2 at address 0x68
I2C error 2 at address 0x69
I2C error 2 at address 0x6A
I2C error 2 at address 0x6B
I2C error 2 at address 0x6C
I2C error 2 at address 0x6D
I2C error 2 at address 0x6E
I2C device found at address 0x6F
I2C error 2 at address 0x70
I2C error 2 at address 0x71
I2C error 2 at address 0x72
I2C error 2 at address 0x73
I2C error 2 at address 0x74
I2C error 2 at address 0x75
I2C error 2 at address 0x76
...
```

but instead we get this:

```
I2C device found at address 0x1
I2C device found at address 0x2
I2C device found at address 0x3
I2C device found at address 0x4
I2C device found at address 0x5
I2C device found at address 0x6
I2C device found at address 0x7
I2C device found at address 0x8
I2C device found at address 0x9
I2C device found at address 0xA
I2C device found at address 0xB
I2C device found at address 0xC
I2C device found at address 0xD
I2C device found at address 0xE
I2C device found at address 0xF
I2C device found at address 0x10
I2C device found at address 0x11
...
```

**Diagnosis:**

Looking at the I2C bus with a logic analyzer, we can see the sub-one cycle clock stretching (note the gap before the NACK):

[image]

The I2C bus traffic is correct and the processor should see this as a NACK (device not present). But ```endTransmission``` is returning zero. It shouldn't...

Here's the correct ACK at address 0x6F:

[image]

Here's the same test at 400kHz. It gives the same false positives. The clock stretching is much more obvious:

[image]

But at 50kHz (thank you v2!), all is well. Note the lack of clock-stretching:

[image]

```
...
I2C error 2 at address 0x6B
I2C error 2 at address 0x6C
I2C error 2 at address 0x6D
I2C error 2 at address 0x6E
I2C device found at address 0x6F
I2C error 2 at address 0x70
I2C error 2 at address 0x71
I2C error 2 at address 0x72
I2C error 2 at address 0x73
...
```

**Suspected cause:**

I think the root cause is our old friend SMPCNT:

[image]

100kHz and 400kHz set SMPCNT to 3:

https://github.com/sparkfun/mbed-os-ambiq-apollo3/blob/be07f057170a4e9ed4a286abb2170b9df3d52de3/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/sdk/mcu/apollo3/hal/am_hal_iom.c#L2203

https://github.com/sparkfun/mbed-os-ambiq-apollo3/blob/be07f057170a4e9ed4a286abb2170b9df3d52de3/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/sdk/mcu/apollo3/hal/am_hal_iom.c#L2223

But the ```default``` setting (which is what would be used for 50kHz), sets SMPCNT to 33 (0x21):

https://github.com/sparkfun/mbed-os-ambiq-apollo3/blob/be07f057170a4e9ed4a286abb2170b9df3d52de3/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/sdk/mcu/apollo3/hal/am_hal_iom.c#L2243

Now, I don't know what the answer to this is... I've tried manually setting SMPCNT for 100kHz and I can't find a value which works. There is a cryptic note in the Apollo3 Blue datasheet:

[image]

so maybe only certain values work for SMPCNT which "allow for synchronization time"?

PaulZC, 2021-06-01:
Possibly also seen here with u-blox GNSS modules. @paulvha suspects SMPCNT too:

https://forum.sparkfun.com/viewtopic.php?f=168&t=55659&p=225605&hilit=smpcnt#p225605

Wenn0101, 2021-07-08:
I have tried quiet a few things to try to get to the root of this.

I agree the datasheet is quiet cryptic - offering very little to help.
A couple interesting observations, that I'm leaving as a note for myself:
-The clock timings after the clock stretch are correct, indicating that something about the processor is correctly determining that the clock stretching is happening (and that the subsampling set by SMPCNT is doing something right...ish)
-You can disable the ability to detect clock stretching smaller than 1 clock cycle (STRDIS), in my testing at 100khz this actually works substantially better!
-The problem doesn't appear to be limited to sub 1 clock cycle stretching (see the 400khz above), but perhaps all clock stretching in general.
-The status of a NAK is kept in the INTSTAT register. Observing this register shows that the non-clock stretched example properly shows a NAK when no ACK occurs, however the result is different with clock stretching. This indicates the problem is in the hardware config or transmission firmware, and NOT in the error handling in the software after the write.

I have updated the 100khz config to disable the sub 1 cycle clock stretching for now, since it seems to work significantly better than having it enabled, but looking at the scope of the clock after this, I am not confident that this is any kind of permanent solution (the last clock has a very inconsistent timing).

PaulZC, 2021-07-08:
Nice one - thank you!
Paul

PaulZC, 2021-07-09:
Hi @Wenn0101 ,
Humm. I think I have bad news here...
When I run the new OpenLog Artemis code on v2.1.1 and connect it to a u-blox ZED-F9P at 100kHz, I see _way_ more I2C checksum errors than with v2.1.0.
I will try and pin this down a bit more. I'll try 400kHz on v2.1.1 and see what that gives me.
Cheers for now,
Paul

Wenn0101, 2021-07-13:
Yea, the solution I found was a bit of a hack. I suppose I shouldn't be surprised that it didn't work well with all devices. I need to find a better solution to this problem, and pull the change that I made in the next release and hopefully i'll have a better solution in the next one.

paulvha, 2021-07-14:
After reading I have done a number of tests to see whether I can help.
**My conclusion in short : the issue only happens when there is a (very) short clock stretch**.

I have used an SCD30, which is using clock stretch, and the code example from PaulCZ to detect. The SCD30 clock stretch is not consistent in timing, but is always longer than the SCL sample cycle. I don't own a QWIIC button or ZED-F9P. The SCD30 was ALWAYS detected correctly no matter what speed

**Configuration**
I have checked the IOM I2C configuration setting between V1.2 and V2.1.1. Apart from the change made by Wenn010 for 100Khz, they are the same

**Timing**
With the SCD30 either on 100KHZ or 400KHZ, I see the clock cycle timing is really 100KHZ or 400KHZ whether 1.2.1 or 2.1.1.

BUT the time between the sending the different I2C addresses is about 0.1mS with V1.2.1, where it is about 3mS with V2.1.1. This is the result of the HUGE overhead of MBED. 

**Clock stretch**
On the scope output, I clearly see clock stretch happening, but always longer than one SCL clock cycle:

V1.2.1 / 100Khz
[image]

V2.1.1 / 100Khz
[image]

V1.2.1 / 400Khz
[image]

V2.1.1 / 400Khz
[image]

jerabaul29, 2021-08-20:
Can this be related to the problems encountered when trying to communicate with the BNO08x?

PaulZC, 2021-08-20:
Hi JR,

Yes, almost certainly.

Best wishes,
Paul

paulvha, 2021-12-13:
I think I have found the root cause and the solution. I bought a QWIIC RED button board a week ago and started deep-dive on the I2C-IOM settings. 

### Clock stretching
The good news is that **clock stretching is working perfectly** well an Apollo3. No need to change anything for that from the default values. 

### ALLWAYS ACK
The real issue is that the Apollo3 has a problem when only the address is sent (as is done with the I2C scanner) to a device that is **performing clock stretching**, the Apollo3 will **ALWAYS report back an ACK**, even if the logic analyzer clearly shows it should be a NAK. _**BUT when sending the address + a dummy byte (something like 0x21) it will detect the NAK and ACK 100% correct.**_  The adjusted sketch is in Appendix B in the attached document.

### QWICC button
I have also done a deep-dive on the QWIIC button hardware and code. It is amazing to see the clock stretch happening there . I have documented the reasons in the attached document.

###  I2C speed
The other aspect that PaulCZ observed **"Actual clock speed via 'scope is 267kHz**" on a 400Khz speed is caused by an incorrect SCLENDLY setting of 3. Do not make this higher than 2 and you have 400Khz.
Also, I would recommend changing the setting for 1Mhz as the current setup brings 857Khz and it could be 920Khz with a better duty cycle.

The attached document is full of the findings of the deep-dive / compares, screenshots from the logic analyzer and  more information about I2C settings and insight in the QWIIC button.

[I2C_clock_stretch.odt](https://github.com/sparkfun/Arduino_Apollo3/files/7705145/I2C_clock_stretch.odt)

hope this helps

regards,
Paul

PaulZC, 2021-12-14:
Excellent detective work Paul (@paulvha )!
Sincere thanks,
Paul (@PaulZC )

Wenn0101, 2021-12-14:
Thanks for the detailed analysis! 
I will read thru this full report today, and see what changes I need to make.

-Kyle