sparkfun/Arduino_Apollo3 #402: Core V1.2.2 - Error Compiling RTC Examples

nigelb, 2021-06-10:
### Subject of the issue
It would appear that when `Apollo3RTC` was renamed `APM3_RTC` [`Apollo3RTC::clearInterrupt`](https://github.com/sparkfun/Arduino_Apollo3/blob/9f309a09712637a58e334981d000e1ff78aa575e/libraries/RTC/src/RTC.cpp#L216) was missed:

### Your workbench
> * What platform are you using?

OS: Ubuntu 20.04.2 LTS and Win10
Arduino: 1.8.13
sparkfun/Arduino_Apollo3: 1.2.2

> * What version of the device are you using? Is there a firmware version?

[SparkFun RedBoard Artemis ATP](https://www.sparkfun.com/products/15442)

### Steps to reproduce

1. With Arduino and and Arduino_Apollo3 added as per these [instructions](https://github.com/sparkfun/Arduino_Apollo3#arduino-boards-manager-recommended) install Sparkfun Apollo3 Boards version 1.2.2 and select one of the Apollo3 boards.
2. Open the Example1_Get_Time sketch.
3. Press the compile button.

### Expected behaviour
The firmware should compile.

### Actual behaviour

```
/opt/arduino/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /opt/arduino/arduino-1.8.13/hardware -hardware /home/user/.arduino15/packages -tools /opt/arduino/arduino-1.8.13/tools-builder -tools /opt/arduino/arduino-1.8.13/hardware/tools/avr -tools /home/user/.arduino15/packages -built-in-libraries /opt/arduino/arduino-1.8.13/libraries -libraries /home/user/Arduino/libraries -fqbn=SparkFun:apollo3:amap3atp:svl_baud=921600,loader=sparkfun_svl -vid-pid=1A86_7523 -ide-version=10813 -build-path /tmp/arduino_build_366060 -warnings=none -build-cache /tmp/arduino_cache_926083 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/user/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major -prefs=runtime.tools.arm-none-eabi-gcc-8-2018-q4-major.path=/home/user/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major -verbose /home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/libraries/RTC/examples/Example1_Get_Time/Example1_Get_Time.ino
/opt/arduino/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /opt/arduino/arduino-1.8.13/hardware -hardware /home/user/.arduino15/packages -tools /opt/arduino/arduino-1.8.13/tools-builder -tools /opt/arduino/arduino-1.8.13/hardware/tools/avr -tools /home/user/.arduino15/packages -built-in-libraries /opt/arduino/arduino-1.8.13/libraries -libraries /home/user/Arduino/libraries -fqbn=SparkFun:apollo3:amap3atp:svl_baud=921600,loader=sparkfun_svl -vid-pid=1A86_7523 -ide-version=10813 -build-path /tmp/arduino_build_366060 -warnings=none -build-cache /tmp/arduino_cache_926083 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/user/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major -prefs=runtime.tools.arm-none-eabi-gcc-8-2018-q4-major.path=/home/user/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major -verbose /home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/libraries/RTC/examples/Example1_Get_Time/Example1_Get_Time.ino
Using board 'amap3atp' from platform in folder: /home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2
Using core 'arduino' from platform in folder: /home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2
Detecting libraries used...
/home/user/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/arm-none-eabi-g++ -DPART_apollo3 -DAM_PACKAGE_BGA -DAM_PART_APOLLO3 -c -g -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -fdata-sections -Os -ffunction-sections -std=gnu++11 -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m4 -DF_CPU=48000000L -DARDUINO=10813 -DARDUINO_AM_AP3_SFE_BB_ARTEMIS_ATP -DARDUINO_ARCH_APOLLO3 -DPART_apollo3 -DAM_PACKAGE_BGA -DAM_PART_APOLLO3 -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/variants/redboard_artemis_atp/config -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/variants/redboard_artemis_atp/bsp -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/ard_sup -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/ard_sup/ard_supers -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/variants/redboard_artemis_atp -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/am_sdk_ap3/mcu/apollo3 -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/am_sdk_ap3/mcu/apollo3/hal -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/am_sdk_ap3/mcu/apollo3/regs -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/am_sdk_ap3/utils -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/am_sdk_ap3/devices -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/am_sdk_ap3/CMSIS/AmbiqMicro/Include -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include /tmp/arduino_build_366060/sketch/Example1_Get_Time.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for RTC.h: [RTC@1.2]
ResolveLibrary(RTC.h)
  -> candidates: [RTC@1.2]
/home/user/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/arm-none-eabi-g++ -DPART_apollo3 -DAM_PACKAGE_BGA -DAM_PART_APOLLO3 -c -g -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -fdata-sections -Os -ffunction-sections -std=gnu++11 -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m4 -DF_CPU=48000000L -DARDUINO=10813 -DARDUINO_AM_AP3_SFE_BB_ARTEMIS_ATP -DARDUINO_ARCH_APOLLO3 -DPART_apollo3 -DAM_PACKAGE_BGA -DAM_PART_APOLLO3 -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/variants/redboard_artemis_atp/config -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/variants/redboard_artemis_atp/bsp -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/ard_sup -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/ard_sup/ard_supers -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/variants/redboard_artemis_atp -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/libraries/RTC/src -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/am_sdk_ap3/mcu/apollo3 -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/am_sdk_ap3/mcu/apollo3/hal -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/am_sdk_ap3/mcu/apollo3/regs -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/am_sdk_ap3/utils -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/am_sdk_ap3/devices -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/am_sdk_ap3/CMSIS/AmbiqMicro/Include -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include /tmp/arduino_build_366060/sketch/Example1_Get_Time.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/home/user/.arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/arm-none-eabi-g++ -DPART_apollo3 -DAM_PACKAGE_BGA -DAM_PART_APOLLO3 -c -g -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -fdata-sections -Os -ffunction-sections -std=gnu++11 -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m4 -DF_CPU=48000000L -DARDUINO=10813 -DARDUINO_AM_AP3_SFE_BB_ARTEMIS_ATP -DARDUINO_ARCH_APOLLO3 -DPART_apollo3 -DAM_PACKAGE_BGA -DAM_PART_APOLLO3 -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/variants/redboard_artemis_atp/config -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/variants/redboard_artemis_atp/bsp -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/ard_sup -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino/ard_sup/ard_supers -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/arduino -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/variants/redboard_artemis_atp -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/libraries/RTC/src -I/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/cores/ar [... cut ...] SparkFun/hardware/apollo3/1.2.2/libraries/RTC/src/RTC.cpp -o /tmp/arduino_build_366060/libraries/RTC/RTC.cpp.o
/home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/libraries/RTC/src/RTC.cpp:216:6: error: 'Apollo3RTC' has not been declared
 void Apollo3RTC::clearInterrupt() {
      ^~~~~~~~~~
Using library RTC at version 1.2 in folder: /home/user/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/libraries/RTC 
exit status 1
Error compiling for board SparkFun RedBoard Artemis ATP.

```

adamgarbo, 2021-06-10:
Thanks @nigelb,

Also noted here: https://github.com/sparkfun/Arduino_Apollo3/discussions/397#discussioncomment-772624

Wenn0101, 2021-06-10:
Thanks for the reminder!
[v1.2.3 pre-release](https://github.com/sparkfun/Arduino_Apollo3/releases/tag/v1.2.3) created with a fix for this this.

adamgarbo, 2021-07-16:
Hey @nigelb, this issue was resolved by @Wenn0101 in 87a33ce and released in v1.2.3.

closed 2021-07-20 by Wenn0101

nigelb, 2021-07-20:
Yep, compiling for me.