sparkfun/Arduino_Apollo3 #403: Using Serial1 TX as a GPIO (ATP) result in undefined behavior

Wenn0101, 2021-06-10:
#define LED_BLINK 24

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BLINK, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BLINK, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BLINK, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}

No blink on pin 24

Wenn0101, 2021-06-10:
gpio_mode call ap3_hal_gpio_pinconfig_partial
when this function is used, a flag is used to designate what elements of am_hal_gpio_pincfg_t are actionable.
the flag for uFuncSel is not being set resulting in this behavior.

PaulZC, 2021-06-17:
(Sorry Kyle!)
I don't have complete proof on this - or a simple "how to replicate" example - but I am having similar grief trying to manually configure the pull-up on the SPI CIPO pin D6 for OpenLog Artemis. In v1.2.1 we could call ```padMode``` to manually set the pull-up on D6 and that worked just fine. But in v2.1 ```padMode``` is no longer supported. I've switched to ```pin_config``` but I don't seem to be able to change the pull-up. Interestingly, I _can_ change the pull-ups on D8 and D9 (the OLA uses those for "qwiic" I2C communication). I just can't seem to be able to do the same on D6 for CIPO.
I suspect that I will need to ask you to set the flag for ```.ePullup``` to say that the pull-up is actionable too on that pin. I just can't prove it yet.
Watch this space, I hope to be able to give you a "how to replicate" example tomorrow.
Sincere thanks,
Paul

Wenn0101, 2021-07-20:
fixed this issue in v2.1.1

closed 2021-07-20 by Wenn0101