sparkfun/Arduino_Apollo3 #406: Artemis Module target doesn't compile

Wenn0101, 2021-06-10:
This target was added in the initial release of v2.0.0, but was never given the attention to fully work.

I doing this now.

The goal is to define as little about the target as possible so that a user who wants to make their own custom board can quickly do so.

So far I have only defined 3 things.
USB_SERIAL_RX 49
USB_SERIAL_TX 48 
LED1/LED_BUILTIN 5

I believe all of these to be required to make an mbed library and arduino core. They could also assist someone bringing up a board for the first time and allow them to test code on a custom board and and ATP for side by side comparisons. A user may want to change these and a recompile of the mbed library would be required, but this would be also be true if these values were left blank for a user to define themselves anyways.

All other SPI, UART, and Wire interfaces are left undefined for the user to define as they desire.

PaulZC, 2021-06-16:
Just a note that the ```PinNames.h``` for the Module is missing a comma after ```D9 = 9```:

https://github.com/sparkfun/mbed-os-ambiq-apollo3/blob/be07f057170a4e9ed4a286abb2170b9df3d52de3/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/TARGET_SFE_ARTEMIS_MODULE/PinNames.h#L47

PaulZC, 2021-06-16:
Ah, oops, I guess I'm only seeing this because I'm using the ```release_candidate``` branch of Apollo3. Looks like this is already fixed in the main branch...

Wenn0101, 2021-06-16:
It should be fixed on release candidate. Try running `git submodule update --init --recursive`

PaulZC, 2021-06-17:
That fixed it. Thank you!
(I shall bookmark this! ;-) )

closed 2021-07-20 by Wenn0101