sparkfun/Arduino_Apollo3 #409: cannot use the DS18B20 with the core v 1.2.1 / OneWire problem?

jerabaul29, 2021-07-02:
I am using the core 1.2.1 and I cannot change that (because of some legacy code that would take a lot of time to re-write).

I need to add some temperature measurement capability to my design. For this, I have bought some DS18B20 : https://www.sparkfun.com/products/11050 . I do am using a 10kOhm pullup resistor between the signal and 5V pins.

This works perfectly fine on Arduino Uno. For example, running the sketch onWireSearch from the DallasTemperature library, the sensor is well recognized (and by running other sketches I am able to read from it too):

```
// Start oneWireSearch.ino 
//

uint8_t pin2[][8] = {
  {
0x28, 0x08, 0x42, 0x8D, 0x0C, 0x00, 0x00, 0x2A  },
};
// nr devices found: 1

//
// End oneWireSearch.ino 
//
```

However I have been trying for several hours to get this to work on the Artemis Redboard, without success. Does anybody has any experience / has had more luck with that? It looks like the problem comes from not being able to talk to the OneWire device at all, any experience with OneWire on Artemis?

jerabaul29, 2021-07-02:
I actually gets this warning when compiling on the Artemis, any other people with experience of it?

```
In file included from /home/jrmet/Arduino/libraries/OneWire/OneWire.cpp:144:
/home/jrmet/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:416:2: warning: #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture." [-Wcpp]
 #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture."
  ^~~~~~~
```

jerabaul29, 2021-07-02:
Ok, that seems to be a library issue. I started trying all libraries one after the other, and it seems that the DS18x20_Temperature example from the One Wire library does work. Very strange, I do not understand why one example works while the other example (that actually uses the exact same onewire library under the hood) does not. Not sure either why there is a problem with the Artemis platform but not the Arduino Uno. Could this be a pin matching issue or something like this?

jerabaul29, 2021-07-02:
Ok, now I actually think that my board had a damaged pin. It looks like actually both libraries work if I use pin 10, and no library works if I use pin 2. I think this can indicate either a mismatch in the pin mapping (unlikely), or more likely, that I have a dead pin. Will try with another board.

jerabaul29, 2021-07-02:
Ok, this is actually very strange. I grabbed another Artemis Redboard, and same observation: I cannot detect the DS18B20 when it is on pin 2 with neither of the examples, but on pin 10, all works fine. I would be surprised if I had 2 separate boards with both having a dead pin 2 and a fine pin 10.

- is anyone else able to reproduce?
- any idea where this comes from?

jerabaul29, 2021-07-02:
BI did a small test. On both of my boards, the DS18B20 works fine on pins 8, 9, 10, but does not work at all on pins 2 to 7 included. Very strange. Redboard, core 1.2.1.

jerabaul29, 2021-07-02:
I was using a 10kOhm following some discussions on the product page. But actually reading the datasheet, the recommended value is 4.7kOhm pullup. Using a 4.7kOhm pullup, all pins work, also pins 2 to 7 :) .

# conclusion

There is somehow something different regarding input impedance / pullup pulldown settings between pins 2-7 vs 8-10.

jerabaul29, 2021-07-02:
I keep this open for now, as this may point to some weird resistors settings. But feel free to close whenever fits you :) .

jerabaul29, 2021-07-02:
I am hitting so many problems with these specific DF18B20 that I tried some from Adafruit. Worked out of the box. I had bought the "bad" DS18B20 from Sparkfun. Wonder if Sparkfun has received some bad quality DS18B20.

jerabaul29, 2021-07-02:
Ok, so this is actually a bit more mixed. I have some Adafruit sensors that work and some that do not. Wonder if there is something more fundamental going on than just good or bad sensors.

I am completely unable to use several of the sensors on the Artemis Global Tracker (though I manage to use the exact same sensors on the readboard), which is my aim. I had opened an issue on the AGT repo but @PaulZC prefers that we discuss it here, so good for me :) .

I must say I am very confused here.

PaulZC, 2021-07-02:
Hi JR,
Can I suggest hanging a 'scope on the data line. Do you see any difference in the max/min voltage levels or the signal rise/fall times between a 'working' chip and a 'non-working' chip? And/or a 'working' pin vs. a 'non-working' pin?
Maybe the pin pull-up needs to be just right?
Best wishes,
Paul

jerabaul29, 2021-07-02:
Unfortunately I do not have an oscilloscope available :( .

jerabaul29, 2021-07-02:
I find it quite strange that the pullup value would need to be exactly right for this to work - would not that be really finicky? Also, I am now using the value recommended in the datasheet.

Also a bit puzzled:

- why is it that the result depend on which pin I use on the Redboard (i.e. pin 2-7 are not ok with the 10k pullup that was recommended on a discussion, while 4.7k works; by contrast, both 4.7k and 10k work on the ping 8-10)

- how is it possible that a sensors that works on the Redboad does not work with the AGT.

jerabaul29, 2021-07-02:
I looked a bit in more details at the code pointed to by the compiler warning. It points to this file I think:

https://github.com/PaulStoffregen/OneWire/blob/master/util/OneWire_direct_gpio.h

If I understand well, the author of the OneWire library has defined 'direct digital read and write' macros and functions, probably to go faster than the functions traditional from the Arduino IDE.

Is it possible that the Artemis platform would need to custom define its own set of such macros for these sensors to work reliably with the Artemis?

jerabaul29, 2021-07-02:
I had a look at the way GPIO are done in the core v1. If I understand correctly this comes from:

https://github.com/sparkfun/Arduino_Apollo3/blob/v1/cores/arduino/ard_sup/gpio/ap3_gpio.cpp

I am not sure why this would not be fast enough or similar for the needs here. Will try to look at it more in the days to come, but I am quite n00b on such low level topics.

jerabaul29, 2021-07-09:
I have had a look at this but I am not very confident about doing this kind of low level stuff. Would be very helpful if we had first class support / robust one wire protocol in the core v1. Anybody with some experience with the low level layers of the core who could help on this? :) .

jerabaul29, 2021-08-02:
Update: I finally had the time to look into this in details. I can now get this to work with both the AGT too :) . I have been doing tests on the AGT only today, so all of the following is for the AGT, not the redboard (seems that values are different, since I before this point got it to work with the redboard, but not the AGT).

To summarize, on the AGT:

- using 1 temperature sensor, with 3.3V input, works with pullups between 22kOhm and 82kOhm. 18kOhm does not work. 100kOhm does not work
- using 2 temperature sensors, with 3.3V input, works with pullups between 10kOhm and 56kOhm. 8.2kOhm does not work. 68kOhm does not work

I will test with 3 and 4 sensors too, as this is for building a "home made thermistor string", and I will let you know how it looks.

Really weird, as these values are far off compared with what would be expected from the datasheet and the online recommendations (typically 4.7kOhm) if I understand well. Really puzzled by this, and the fact that the values that work are not the same on the Redboard, vs the AGT, vs "normal" arduinos. Any idea what the cause could be @PaulZC ? :) .

jerabaul29, 2021-08-02:
With 6 sensors, I find that I need some pullups with value 2.7K to 18K. So the value of the pullups seems to definitely change with the number of sensors used (guess this is not so surprising).

EvalityCCA, 2021-10-03:
Hi jerabaul29

Did you manage to get OneWire working on an Artemis board?
Do you have a working version of the OneWire library for it?
I am similarly trying to get a single DS18B20 working on Artemis with no luck.

jerabaul29, 2021-10-03:
Hi @EvalityCCA ,

Yes, I got it to work, without changing anything, by tweaking the value of the pullup resistors, see the comment above from Aug. 2 :) .

EvalityCCA, 2021-10-03:
Thanks for the quick response!

I tried tweaking the pullup resistors but without any luck.
I get compile issues with the standard OneWire library due to the interrupts() and noInterrupts() that aren't declared for the Artemis, so I'm using an untested version of this library.
How did you get around this issue or which OneWire library did you use?

jerabaul29, 2021-10-03:
I just used the default Wire library on the core v1, got no compile issue there.

EvalityCCA, 2021-10-03:
Thanks again for the quick response.

jerabaul29, 2022-01-09:
Mmmh, actually still having some issues with the DS18B20 and artemis. I do not have access with an oscilloscope, if anyone who has one want to look at this one day would be amazing.

jerabaul29, 2022-01-10:
A small note: I started trying to tweak the timing of the OneWire library. Seems like reducing the delayMicroseconds to be on the "shorter delays" side of the spec for the 1 Wire protocol does help a lot. Wonder if 1) this may be because going through the Arduino API digital read / write / pin setup may be slow, so this may add duration there, or if 2) there may be a prefactor wrong somewhere and the delayMicroseconds actually waits longer than what it says (? wondering this since it seems the threshold for getting the OneWire to work is about actualDelayMicrosecond = specMaxDelayMicrosecond / 4 ). Anyways, seems like this works quite fine now, crossing fingers this holds in tests:

https://github.com/jerabaul29/OneWire/tree/feat/Artemis_compatible

jerabaul29, 2022-01-10:
(conclusion: I do need an oscilloscope or a logical bus analyzer badly, otherwise this is just guessing and shooting in the dark...).

jerabaul29, 2022-01-10:
This is now described in much more details / solved at: https://github.com/paulvha/apollo3/tree/master/OneWire_on_uart . Many thanks @paulvha :) .

closed 2022-01-10 by jerabaul29

NezhaZahri, 2022-04-13:
@jerabaul29 
So I've tried to use the OneWire library from PaulStoffregen on the Artemis Global Tracker.
It seems that I have the same error as you.
`In file included from C:\Users\nezha\OneDrive\Documents\Arduino\libraries\OneWire-2.3.6\OneWire.cpp:144:
C:\Users\nezha\OneDrive\Documents\Arduino\libraries\OneWire-2.3.6\util/OneWire_direct_gpio.h:459:2: warning: #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture." [-Wcpp]
 #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture."
`
How did you get it up and running?
I tried to run the example of [DS18x20_Temperature](https://github.com/PaulStoffregen/OneWire/tree/master/examples/DS18x20_Temperature) as well.
What other libraries did you try?