sparkfun/Arduino_Apollo3 #415: Arduino BLE with watchdog interrupt deep sleep

blasteroid297, 2021-07-09:
Hello

I have a project I'm working on that I want to turn on BLE for a quick interval and connect to my phone. If there is no connection after a short time then it should just go back into deep sleep. After this, I wanted watchdog to wake the system back up after 10 seconds and repeat this process. So far, I have used the Arduino BLE library and watchdog deep sleep examples for waking up the system after 10 seconds and that works fine but when I bring in BLE to it, the system constantly goes into deep sleep and immediately wakes up after around half a second instead of 10 seconds. If I comment out the BLE.begin lines, the problem goes away, but obviously BLE is not on anymore.

I'm not sure if there is a bug between using BLE.begin and watchdog timers together.

This was the deep sleep watchdog example I started with:
[https://github.com/sparkfun/Arduino_Apollo3/blob/main/libraries/WDT/examples/Example3_WDT_LowPower/Example3_WDT_LowPower.ino]

I have already talked about this issue on sparkfun's [forum](https://forum.sparkfun.com/viewtopic.php?f=169&t=56008) 

How do you utilize watchdog with BLE to wake up from deep sleep every some user-defined interval?

Perhaps there could be a dedicated WDT with BLE example to showcase this.
I am very new to this, so I don't know if I'm missing something simple.

Wenn0101, 2021-07-20:
Could you post the sketch you are using that has the issue? It would be great to have that as reference point.

If I could try to summarize your issue, it would be that you cannot get the device to enter sleep when the sketch uses BLE?

blasteroid297, 2021-07-21:
Well I think something other than watchdog is waking up the system from deep sleep immediately. I don't know what that is exactly. It could be BLE waking up the system?(even though I think I turned it off before going into deep sleep).

It's not that I can't go into deep sleep. That part works. It's that the system wakes from deep sleep way earlier than the 10 seconds that I had set with the watchdog.

My code is basically just an edited version of the [Example3_WDT_LowPower.ino](https://github.com/sparkfun/Arduino_Apollo3/blob/main/libraries/WDT/examples/Example3_WDT_LowPower/Example3_WDT_LowPower.ino) in arduino_apollo3 repository where I attempted to turn BLE on and off after deep sleep cycles. 

Here is my arduino code:
`#include "RTC.h"
#include "WDT.h"
#include <ArduinoBLE.h>

BLEService motorService("19B10000-E8F2-537E-4F6C-D104768A1214"); // create service

// create motor characteristic and allow remote device to write
BLEByteCharacteristic motorCharacteristic("19B10001-E8F2-537E-4F6C-D104768A1214", BLEWrite);
int flag = 0;

volatile bool alarmFlag = false; // RTC ISR flag
volatile bool watchdogFlag = true; // Watchdog Timer ISR flag
volatile int watchdogInterrupt = 0; // Watchdog interrupt counter

void setup()
{
  Serial.begin(115200);

  Serial.println("Artemis Watchdog Low Power Example");

  // Set the RTC time using UNIX Epoch time
  rtc.setEpoch(1596240000); // Saturday, August 1, 2020 00:00:00

  // Set the RTC's alarm
  //rtc.setAlarm(0, 0, 0, 0, 0, 0); // Set alarm (hund, ss, mm, hh, dd, mm)
  //rtc.setAlarmMode(6); // Set the RTC alarm to trigger every minute
  //rtc.attachInterrupt(); // Attach RTC alarm interrupt

//  Serial.print("Before BLE: ");
//  BLEsetup();
//  Serial.print("After BLE: ");
//  printDateTime();
  am_hal_ble_state_t *pBle = pHandle;
  am_hal_ble_wakeup_set(pBle, 0);
  // Configure the watchdog timer
  // See Example2_WDT_Config for more information on how to configure the watchdog
  wdt.configure(WDT_16HZ, 160, 240); // 16 Hz clock, 10-second interrupt period, 15-second reset period

  // Start the watchdog
  wdt.start();
}
/////////////Start Bluetooth and start advertising
void BLEsetup() {
  //Serial.begin(9600);
  //while(!Serial);

  pinMode(0, OUTPUT);
  pinMode(1, OUTPUT);
  
  if (!BLE.begin()) {
    //Serial.println("starting BLE failed!");
    while(1);
  }
  Serial.println("BLE Begun");
  BLE.setLocalName("MotorCallback");
  BLE.setAdvertisedService(motorService);
  motorService.addCharacteristic(motorCharacteristic);
  BLE.addService(motorService);

  //BLE.setEventHandler(BLEConnected, blePeripheralConnectHandler);
  //BLE.setEventHandler(BLEDisconnected, blePeripheralDisconnectHandler);

  // assign event handlers for characteristic
[... 117 lines cut ...]

  // Enable Serial
  Serial.begin(115200);
  Serial.println("Woke Up!");
  printDateTime();
}

// Interrupt handler for the watchdog.
extern "C" void am_watchdog_isr(void)
{
  // Clear the watchdog interrupt
  wdt.clear();

  // Perform system reset after 10 watchdog interrupts (should not occur)
//  if ( watchdogInterrupt < 10 )
//  {
//    //wdt.restart(); // "Pet" the dog
//  }
//  else {
//    while (1); // Wait for reset to occur
//  }

//  watchdogFlag = true; // Set the watchdog flag
//  watchdogInterrupt++; // Increment watchdog interrupt counter
}`

nigelb, 2022-07-13:
Greetings @blasteroid297 and @Wenn0101,

I too have run into the issue of the BLE stack breaking my deep sleep.

For the initial proof of concept I am using **Arduino 1.8.13** and **sparkfun/Arduino_Apollo3 2.2.1** on **Ubuntu 20.04.4 LTS**

I started with [Example6_LowPower_Alarm.ino](https://github.com/sparkfun/Arduino_Apollo3/blob/main/libraries/RTC/examples/Example6_LowPower_Alarm/Example6_LowPower_Alarm.ino) and made the following changes: 

```diff
--- Example6_LowPower_Alarm.ino.orig	2022-07-12 16:17:06.394151957 +1000
+++ Example6_LowPower_Alarm.ino	2022-07-13 11:16:04.112499202 +1000
@@ -17,12 +17,20 @@
 */
 
 #include "RTC.h"
+#include "ArduinoBLE.h"
+
+bool ble = true;
 
 void setup()
 {
   Serial.begin(115200);
   Serial.println("SparkFun RTC Low-power Alarm Example");
 
+  if(ble){
+    BLE.begin();
+    BLE.end();
+  }  
+
   // // Easily set RTC using the system __DATE__ and __TIME__ macros from compiler
   // RTC.setToCompilerTime();

```
to arrive at the following:

```c++
/*
  Author: Adam Garbo and Nathan Seidle
  Created: June 3rd, 2020

  This example demonstrates how to set an RTC alarm and enter deep sleep.

  The code is configured to set an RTC alarm every minute and enter
  deep sleep between interrupts. The RTC interrupt service routine will
  wake the board and print the date and time upon each alarm interrupt.

  Tested with a SparkFun Edge 2. Confirmed sleep current of 2.5 uA.
*/

/* 
// This file is subject to the terms and conditions defined in
// file 'LICENSE.md', which is part of this source code package.
*/

#include "RTC.h"
#include "ArduinoBLE.h"

bool ble = false;

void setup()
{
  Serial.begin(115200);
  Serial.println("SparkFun RTC Low-power Alarm Example");

  if(ble){
    BLE.begin();
    BLE.end();
  }  

  // // Easily set RTC using the system __DATE__ and __TIME__ macros from compiler
[... 330 lines cut ...]
  }
```

with this:

```c++
  if(ble){
    BLE.begin();
    BLE.end();
    am_hal_ctimer_stop(7, AM_HAL_CTIMER_TIMERB);
  }
```

then compile and run this and the output is:

```
SparkFun RTC Low-power Alarm Example
Alarm interrupt: 2020-6-3 12:59:50.0
⸮Alarm interrupt: 2020-6-3 13:0:0.1
```
and a power trace of:

[image]

In summary, it seems that the code called by `BLE.begin();` starts Timer 7B and then `BLE.end();` fails to stop it once everything is done.