sparkfun/Arduino_Apollo3 #417: Compilation of most Arduino libraries fails for LoRa Thing Plus Explorable

amirna2, 2021-07-11:
### Subject of the issue
I am testing several very common Arduino Libraries such as ArduinoJSON or the Adafruit sensor Libraries. They all fail with various compilation errors when using the SparkFun Apollo3 Arduino core. I am currently testing with the LoRa explorable board.

### Your workbench
* What platform are you using?
  * Macbook Pro - OSX (Big Sur 11.4) - Arduino IDE 1.8.15
* What version of the device are you using? Is there a firmware version?
  *  LoRa Things Plus Explorable - Arduino Core 2.1.1
* How is the device wired to your platform?
* How is everything being powered?
  * USB connection to the MackBook Pro
* Are there any additional details that may help us help you?
  * I am testing an example provided by ArduinoJSON library version 6.16.1
  * I validated my development setup with the example provided with the Appolo3 board
    * Tested Example10_DisplayICRevision which shows board revision `B0`
  
### Steps to reproduce
* Load and install ArduinoJSON library from the Arduino IDE (I used version 6.16.1)
* Open the `StringExample.ino`
* Select `LoRa Thing Plus Explorable` as your board

### Expected behaviour
* The `StringExample.ino` should compile successfully 
### Actual behaviour
* The `StringExample.ino` fails to compile with the following errors

```
Compiling sketch...
/Users/anathoo/Library/Arduino15/packages/SparkFun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/arm-none-eabi-g++ -include /Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/variants/LoRa_THING_PLUS_expLoRaBLE/mbed/mbed_config.h -include /Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/cores/arduino/sdk/ArduinoSDK.h -iprefix /Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/cores/ @/Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/variants/LoRa_THING_PLUS_expLoRaBLE/mbed/.cxx-flags -MMD -DARDUINO=10815 -DARDUINO_LoRa_THING_PLUS_expLoRaBLE -DARDUINO_ARCH_MBED -DARDUINO_ARCH_APOLLO3 -DMBED_NO_GLOBAL_USING_DIRECTIVE -DCORDIO_ZERO_COPY_HCI @/Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/variants/LoRa_THING_PLUS_expLoRaBLE/mbed/.cxx-symbols -I/Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/cores/arduino -I/Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/variants/LoRa_THING_PLUS_expLoRaBLE -I/Users/anathoo/Documents/Arduino/libraries/ArduinoJson/src -I/Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/cores/arduino -I/Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/cores/arduino/mbed-bridge -I/Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/cores/arduino/mbed-bridge/core-api -I/Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/cores/arduino/mbed-bridge/core-api/api/deprecated @/Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/variants/LoRa_THING_PLUS_expLoRaBLE/mbed/.includes /var/folders/57/0h65vzcn5px_061yp_bmc31r0000gn/T/arduino_build_707631/sketch/StringExample.ino.cpp -o /var/folders/57/0h65vzcn5px_061yp_bmc31r0000gn/T/arduino_build_707631/sketch/StringExample.ino.cpp.o
In file included from /Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/cores/arduino/mbed-bridge/core-api/api/String.h:33,
                 from /Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/cores/arduino/mbed-bridge/core-api/api/Print.h:24,
                 from /Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/cores/arduino/mbed-bridge/core-api/api/Stream.h:25,
                 from /Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/cores/arduino/mbed-bridge/core-api/api/Client.h:22,
                 from /Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/cores/arduino/mbed-bridge/core-api/api/ArduinoAPI.h:29,
                 from /Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/cores/arduino/mbed-bridge/Arduino.h:17,
                 from /Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/cores/arduino/sdk/ArduinoSDK.h:9,
                 from <command-line>:
/Users/anathoo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In instantiation of 'typename ArduinoJson6180_91::enable_if<ArduinoJson6180_91::is_pointer<T>::value, T>::type ArduinoJson6180_91::pgm_read(const void*) [with T = const arduino::__FlashStringHelper*; typename ArduinoJson6180_91::enable_if<ArduinoJson6180_91::is_pointer<T>::value, T>::type = const arduino::__FlashStringHelper*]':
/Users/anathoo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/DeserializationError.hpp:100:12:   required from here
/Users/anathoo/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.1.1/cores/arduino/mbed-bridge/core-api/api/deprecated-avr-comp/avr/pgmspace.h:106:29: error: 'const void*' is not a pointer-to-object type
 #define pgm_read_ptr(addr) (*(const void *)(addr))
                            ~^~~~~~~~~~~~~~~~~~~~~~
Using library ArduinoJson at version 6.18.0 in folder: /Users/anathoo/Documents/Arduino/libraries/ArduinoJson 
exit status 1
Error compiling for board LoRa Thing Plus expLoRaBLE.
```

amirna2, 2021-07-12:
I have a workaround for this issue in `pgmspace.h` where the error is happening
```
//#define pgm_read_ptr(addr) (*(const void *)(addr))
#define pgm_read_ptr(addr) (*reinterpret_cast<const void* const *>(addr))
```
I am suspecting that enforcing c++ with `--language-force=c++` compiler option may be the real issue. Any reason why this option is needed?

Wenn0101, 2021-07-12:
Thanks for the report!

I will add this to my To-Do!

Wenn0101, 2022-03-17:
Issue resolved when arduino api was updated to v1.2.0

closed 2022-03-17 by Wenn0101