sparkfun/Arduino_Apollo3 #420: Issue with PDM examples1_MicrophoneOutput for Sparkfun Edge

Batwho, 2021-07-25:
I could successfully upload the examples1_MicrophoneOutput under the PDM example. However, the output in the Serial Monitor for the loudest frequency is always 0. This issue also happens to example2 and 3. 
[image]
For example 4, the python received data is also always 0. 

I've checked the mic through the micro_speech example provided by the Sparkfun, the one could detect "Yes" and "No", and it works fine.

Batwho, 2021-07-25:
Now I could use Example 4 to get the possible audio data. However, the config is the Edge is running under 500kbps and the python script is receiving data at 115200. If I change the python to 500K or downsample the Edge to 115200, it can not get any data, which is always 0. But for 115200, there is a start spike and then always 0. 
[image]

metanav, 2021-08-17:
I guess Sparkfun edge does not have a PDM mic. It is analog one.

Wenn0101, 2021-09-13:
This is correct. The edge does not have a PDM microphone. This example will work for devices that have a PDM microphone.

closed 2021-09-13 by Wenn0101