sparkfun/Arduino_Apollo3 #422: [library issue]: USB_Host_Shield_2.0 library

Topo-png, 2021-08-21:
### Subject of the issue
Trouble using [USB_Host_Shield_2.0 library](https://github.com/felis/USB_Host_Shield_2.0)

### Your workbench
* RedBoard-Artemis-ATP w/ Sparkfun USB shield 
* Sparkfun board 1.2.1, Arduino 1.8.15
* Device is connected via shield headers
* Device is powered through USB-C on the Artemis ATP board

### Steps to reproduce
I'm attempting to run the "board_qc" example with the end goal of using the XBOXONE controls. I'm trying to troubleshoot why I'm unable to read/write to the max3421 chip. Please see [PR](https://github.com/felis/USB_Host_Shield_2.0/pull/654) for more details.

Wenn0101, 2022-03-17:
Sorry we never really got back to you on this. Did you ever get this resolved/do you still need help? I see the PR is open, was it stalled or just waiting to get pulled in.

Perhaps you could help us out a bit here. Were you seeing a compilation problem, or a problem when actually communicating?
Were you able to debug the communication at all or was their none happening?
I notice you have listed your library version as 1.2.1, have you tried the latest version of this library?

Topo-png, 2022-03-24:
@Wenn0101 I never got this to work properly. I was eventually able to compile and load to the board, but I was never able to run the demo program. [This ](https://github.com/felis/USB_Host_Shield_2.0/pull/654) shows all of the progress I made. I eventually shelved this portion of the project.