sparkfun/Arduino_Apollo3 #424: Artemis Module def fails with I2C libraries

nseidle, 2021-08-31:
When selecting the 'Artemis Module' board the implementation fails to include enough I2C information for a basic I2C sketch to run with a library. Example:

    #include <Wire.h>
    TwoWire myWire(9, 8); //SDA, SCL: Will use pads 9/8

    #include <SparkFun_MCP9600.h> // Click here to get the library: http://librarymanager/All#SparkFun_MCP9600
    MCP9600 tempSensor;

    void setup() {
      Serial.begin(115200);

      myWire.begin();
    }

    void loop() {}

Fails with error 

    ...\Arduino\libraries\SparkFun_MCP9600_Arduino_Library\src/SparkFun_MCP9600.h:100:62: error: 'Wire' was not declared in this scope

because variant.h does not have a `#define VARIANT_WIRE_INTFCS 1` and am_bsp.h does not have an IOM defined. One solution would be to add these but I'm not sure how you want to balance the minimum definition of the 'Artemis Module' board. A user should be able to start whatever peripheral they want and currently that's not possible without modifying the core.

Wenn0101, 2021-08-31:
Ahh i see the problem. Wire isn't available to the SparkFun_MCP9600.h library.

nseidle, 2021-08-31:
Keep going. Your example will compile but as soon as you try to bring in any SparkFun lib, it will fail.

Wenn0101, 2021-08-31:
Hmm, interesting problem. I am not sure the best way to approach it. The easy thing to do would be to just `#define VARIANT_WIRE_INTFCS 1` and define a default Wire.

This isn't what I originally had in mind for the Artemis Module target... but it might be what is necessary.

Wenn0101, 2021-12-14:
This was fixed in the latest release, closing now.

closed 2021-12-14 by Wenn0101