sparkfun/Arduino_Apollo3 #425: linker issues including precompiled library

wemaxon, 2021-09-07:
Hello,

I am having trouble including my custom precompiled static library to my apollo3 project. I am using the Sparkfun Lora Thing Plus. Until a few days ago the linking was working fine until I suddenly started to get "undefined reference" errors to functions and objects in the libmicroros.a library. I did not change the library and I cant figure out why it wont link anymore. 

The library structure looks like this

```
micro_ros
  src
    micro_ros.h
    ...
    cortex-m4
      libmicroros.a
  library.properties
```

And the library.properties file like this

```
name=micro-ros
version=1.0
author=eProsima
sentence=micro-ROS custom compiled static library
paragraph=micro-ROS puts ROS 2 onto microcontrollers
category=ROS 2
url=https://micro.ros.org/docs/tutorials/advanced/create_custom_static_library/
architectures=mbed,apollo3
precompiled=true
ldflags=-lmicroros
```

When compiling i get the following message
```
Compiling libraries...

Compiling library "micro_ros"

Library micro_ros has been declared precompiled:

Using precompiled library in C:\Users\maxan\Documents\Arduino\libraries\micro_ros\src\cortex-m4

The platform does not support 'compiler.libraries.ldflags' for precompiled libraries.
```

The linker then fails with lots of undefined references to functions and objects in the libmicroros.a library
```
"C:\\Users\\maxan\\AppData\\Local\\Arduino15\\packages\\SparkFun\\tools\\arm-none-eabi-gcc\\8-2018-q4-major/bin/arm-none-eabi-gcc" "-TC:\\Users\\maxan\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\apollo3\\2.1.1/tools/uploaders/svl/0x10000.ld" "-Wl,-Map,C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607/SFE_LoRa_Module_Firmware.ino.map" -o "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607/SFE_LoRa_Module_Firmware.ino.axf" -Wl,--whole-archive "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\sketch\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\action_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\action_msgs\\srv\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\actionlib_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\builtin_interfaces\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\composition_interfaces\\srv\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\diagnostic_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\diagnostic_msgs\\srv\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\example_interfaces\\action\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\example_interfaces\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\example_interfaces\\srv\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\geometry_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\lifecycle_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\lifecycle_msgs\\srv\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\micro_ros_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\nav_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\nav_msgs\\srv\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\px4_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\rcl_interfaces\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\rcl_interfaces\\srv\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\rosgraph_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\sensor_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\sensor_msgs\\srv\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\shape_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\statistics_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\std_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\std_srvs\\srv\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\stereo_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\test_msgs\\action\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\test_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\test_msgs\\srv\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\trajectory_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\unique_identifier_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\visualization_msgs\\msg\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\micro_ros\\visualization_msgs\\srv\\detail\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\BasicMAC\\aes\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\BasicMAC\\hal\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\BasicMAC\\lmic\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\libraries\\SPI\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\core\\config\\objs.a" "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\core\\objs.a" -Wl,--no-whole-archive -Wl,--whole-archive "C:\\Users\\maxan\\AppData\\Local\\Temp\\arduino_build_766607\\core\\core.a" -Wl,--no-whole-archive -Wl,--whole-archive "C:\\Users\\maxan\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\apollo3\\2.1.1\\variants\\LoRa_THING_PLUS_expLoRaBLE/mbed/libmbed-os.a" -Wl,--no-whole-archive -Wl,--whole-archive "C:\\Users\\maxan\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\apollo3\\2.1.1/cores/mbed-os/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/sdk/CMSIS/ARM/Lib/ARM/libarm_cortexM4lf_math.a" -Wl,--no-whole-archive "@C:\\Users\\maxan\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\apollo3\\2.1.1\\variants\\LoRa_THING_PLUS_expLoRaBLE/mbed/.ld-flags" --specs=nano.specs --specs=nosys.specs -lsupc++ -lstdc++ -lm -DARDUINO=10816 -DARDUINO_LoRa_THING_PLUS_expLoRaBLE -DARDUINO_ARCH_MBED -DARDUINO_ARCH_APOLLO3 -DMBED_NO_GLOBAL_USING_DIRECTIVE -DCORDIO_ZERO_COPY_HCI "@C:\\Users\\maxan\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\apollo3\\2.1.1\\variants\\LoRa_THING_PLUS_expLoRaBLE/mbed/.ld-symbols"
c:/users/maxan/appdata/local/arduino15/packages/sparkfun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\maxan\AppData\Local\Temp\arduino_build_766607\sketch\objs.a(micro_ros_handler.cpp.o): in function `micro_ros_init()':
micro_ros_handler.cpp:(.text._Z14micro_ros_initv+0x14): undefined reference to `rmw_uros_set_custom_transport'
c:/users/maxan/appdata/local/arduino15/packages/sparkfun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: micro_ros_handler.cpp:(.text._Z14micro_ros_initv+0x22): undefined reference to `rcutils_get_default_allocator'
[... cut ...] _type_support_c.c.obj) does not

c:/users/maxan/appdata/local/arduino15/packages/sparkfun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: failed to merge target specific data of file C:\Users\maxan\Documents\Arduino\libraries\micro_ros\src\cortex-m4\libmicroros.a(libpx4_msgs__rosidl_typesupport_microxrcedds_c-vehicle_local_position__type_support_c.c.obj)
```

I would be very happy if anyone could help solving this
Thanks in advance

Wenn0101, 2021-09-07:
Hello,

It seems that you were able to fix your problem, but then uncovered a new one.

```
SFE_LoRa_Module_Firmware.ino.axf uses VFP register arguments ... libmicroros.a ... does not
```
Is the key clue here, and tells us what is going on.

Our core is built with the `hard` abi, and yours is build with `soft` or `softfp` abi. Looking around it seems like arduino typically uses the `softfp` abi. I will look into changing this, but I believe this was done so that we could better utilize the hardware to perform math in our PDM library.

In the meantime, if you are able to recompile this library with the flags `-mfpu=fpv4-sp-d16 -mfloat-abi=hard` it will work with our core. Let me know if this works and I will look into what it would mean for our core to transition back over to the more typically `softfp` abi so that you wont have to do this in the future.

wemaxon, 2021-09-07:
Hey Wenn0101,

thanks a lot for your quick response.

I recompiled the static library with the flags `-mfpu=fpv4-sp-d16 -mfloat-abi=hard` and my project is linking again. Great!

What I still don't understand is why this problem came up this suddenly. A few weeks ago my project was compiling and linking without any issues. Did something change in Apollo3 Core or the Arduino Toolchain?

closed 2021-09-07 by wemaxon

Wenn0101, 2021-09-07:
this is a good question, and something I was curious about. The apollo3 core changed to the hard abi in around april. Perhaps you just recently updated from an older version of the core?

wemaxon, 2021-09-08:
I see. This explains the second problem. I remember updating the library. Was the Toolchain combine recipe changed too? I remember this was already working even without my change to `platform.txt` at some point.