sparkfun/Arduino_Apollo3 #431: Missing `noInterrupts()` and `interrupts()` functions

santaimpersonator, 2021-11-18:
It appears that the Apollo3 Arduino core doesn't implement these Arduino functions:

* `noInterrupts()`: https://www.arduino.cc/reference/en/language/functions/interrupts/nointerrupts/
* `interrupts()`: https://www.arduino.cc/reference/en/language/functions/interrupts/interrupts/

I believe that enabling and disabling the interrupts were handled by these functions

* `am_hal_interrupt_master_enable()`
* `am_hal_interrupt_master_disable()`

in the Apollo3 Mbed OS platform in these files (location varies by branch):

* [mbed_overrides.c](https://github.com/sparkfun/mbed-os-ambiq-apollo3/blob/ambiq-apollo3-arduino/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/device/mbed_overrides.c)
* [am_hal_interrupt.h](https://github.com/sparkfun/mbed-os-ambiq-apollo3/blob/ambiq-apollo3-arduino/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/sdk/mcu/apollo3/hal/am_hal_interrupt.h#L68-L69)

I saw an [old issue that got pulled](https://github.com/sparkfun/Arduino_Apollo3/pull/44#issuecomment-519674224), where @oclyke mentioned `noInterrupts()` being used; also, there was [an example for the PDM microphone calling those functions](https://github.com/sparkfun/Arduino_Apollo3/pull/44/commits/f42a8cc9c32ae4c18921b0c7f89a7c6d4e1d3f22). However, I think that the `noInterrupts()` and `interrupts()` functions were removed at a later point (probably by accident).

------------------

I found the issue when I was trying to get the [LoRaWAN Arduino library](https://github.com/mcci-catena/arduino-lmic) working with the Artemis MicroMod processor board. However, I ran into compiler issues for `noInterrupts()` and `interrupts()` in the library at these lines:

* https://github.com/mcci-catena/arduino-lmic/blob/master/src/hal/hal.cpp#L340
* https://github.com/mcci-catena/arduino-lmic/blob/master/src/hal/hal.cpp#L346

I was able to work around the issue by throwing an ifdef to avoid compiling those functions, since the disabling of the interrupts in the library seemed mostly critical for AVR boards (from what I could tell).

    void hal_disableIRQs () {
        #if !defined(ARDUINO_APOLLO3_SFE_ARTEMIS_MM_PB)
            noInterrupts();
        #endif // !defined(ARDUINO_APOLLO3_SFE_ARTEMIS_MM_PB)

        irqlevel++;
    }

    void hal_enableIRQs () {
        if(--irqlevel == 0) {
            #if !defined(ARDUINO_APOLLO3_SFE_ARTEMIS_MM_PB)
                interrupts();
            #endif // !defined(ARDUINO_APOLLO3_SFE_ARTEMIS_MM_PB)

wemaxon, 2021-11-24:
I had the exact same problem, which I solved too by commenting out both function calls. In the example provided by SparkFun the calls are removed  as well. I still have some timing issues when trying to receive LoRa frames. I am using the [LacunaSpace BasicMAC](https://github.com/LacunaSpace/basicmac) Library with the SparkFun LoRa Thing Plus explorable.

Wenn0101, 2022-03-17:
Fix is in dev and scheduled for release 2.2.1

closed 2022-03-17 by Wenn0101