sparkfun/Arduino_Apollo3 #432: I2C/Qwiic issues with 2.1.1 and ATP board

sylgeist, 2021-12-06:
Ardunio 1.8.16 on MacOS

I have been out of the loop for a few versions, but after upgrading to 2.1.1 even example sketches fail to communicate over I2C. I can use modules like the Qwiic SerLCD and Qwiic OLED, but Qwiic buttons and LED sticks don't respond.

Running the I2C scanner sketch with one qwiic button attached:
```
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 I2C Scanner
Scanning...
I2C device found at address 0x07  !
I2C device found at address 0x1E  !
I2C device found at address 0x27  !
I2C device found at address 0x6F  !
I2C device found at address 0x72  !
done

Scanning...
I2C device found at address 0x1C  !
I2C device found at address 0x25  !
I2C device found at address 0x2E  !
I2C device found at address 0x37  !
I2C device found at address 0x40  !
I2C device found at address 0x49  !
I2C device found at address 0x52  !
I2C device found at address 0x6F  !
I2C device found at address 0x72  !
I2C device found at address 0x77  !
done

Scanning...
I2C device found at address 0x18  !
I2C device found at address 0x21  !
I2C device found at address 0x2A  !
I2C device found at address 0x33  !
I2C device found at address 0x3C  !
I2C device found at address 0x45  !
I2C device found at address 0x4E  !
I2C device found at address 0x6F  !
I2C device found at address 0x72  !
done

Scanning...
I2C device found at address 0x6F  !
I2C device found at address 0x72  !
done
```

paulvha, 2021-12-06:
Looks the same issue as https://github.com/sparkfun/Arduino_Apollo3/issues/400

sylgeist, 2021-12-06:
Apologies! I did search and must not have had the right key word combo! That does look exactly like what I'm hitting as well.

sylgeist, 2021-12-06:
Am I understanding that the workaround/fix for the moment is to set

`Wire.setClock(400000)`

and what are the implications of doing so? Does that introduce other issues?

sylgeist, 2021-12-06:
Closing as a duplicate

closed 2021-12-06 by sylgeist