sparkfun/Arduino_Apollo3 #438: How TO Speed Up transmission frequency of BLE

JacksonLtt, 2021-12-21:
### Subject of the issue
How could I speed up transmission frequency of BLE?

### Your workbench
* What platform are you using?
  I am using Arduino to program Arduino Apollo3.
* What version of the device are you using? Is there a firmware version?
  I used [SparkFun Thing Plus - Artemi](https://www.sparkfun.com/products/15574)s as microcontroller. 
 Artemi works as a broadcaster to advertise data. 
 I use Macbook Pro (macOS Big Sur, 2018) as a receiver to receive data from Artemi. I used [PythonNUS ](https://github.com/JacksonLtt/sign_language/blob/main/Arduino/sample%20data/PythonNUS-master/prac.py)to test the transmission rate. 

* How is everything being powered?
   I used USB to power the Artemi.

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue to a gist.
 
Firstly, I used [ArduinoBLE/examples/Peripheral/BatteryMonitor/](https://github.com/arduino-libraries/ArduinoBLE/tree/master/examples/Peripheral/one_byte) to test the transmission rate of BLE. The data size is one byte char.
The transmission rate that I got is around **120-130 Hz** (number of packets/per second)

Secondly, I used my [code ](https://github.com/JacksonLtt/sign_language/blob/main/Arduino/Artemis%20Example/BatteryMonitor_example/byte_220)to test the transmission rate of sending 220 byte via BLE. 
The transmission rate that I got is only **30 Hz** (number of packets/per second)

Please let me know how could I speed up transmission rate of BLE

Best regards

paulvha, 2021-12-21:
HI  I am not Sparkfun, but have been working on ArduinoBLE before. Got the driver working on 1.2.1. and looked in-depth at the ArduinoBLE code. Next to I have been working to get working examples on the better MBED-BLE / Cordio stack which I plan to post soon. I saw your one-byte char code. For sure there is a TON of overhead to send just one byte. But the link to your 220 byte is broken. Can you update so I can have a look and maybe help.
regards,
Paul

JacksonLtt, 2021-12-21:
Hello,

Thank you for your reply.
Here is the link of my code for sending 220 bytes.
https://github.com/JacksonLtt/sign_language/tree/main/Arduino/Artemis%20Example/byte_220

Best regards

paulvha, 2021-12-22:
I have reworked the sketches a little bit and have done a test with connecting nRF connect from my phone to an Artemis ATP. I have timed how long it would take for an ATP to send 500 packages using the byte_one and using the byte_220. On the byte_one the average of 10 passes takes about 20,5ms per transmission, on the byte_220 it takes about 18,8ms.  So both are roughly running at 50 transmissions per second.
Not sure how you measured your results. If you can share your test I can try to simulate it.

Maybe a point that is important. The default MTU size is 23 according to BLE standards.  When you create the characteristic with 220 bytes all you do is create a buffer to store the value. When you write / send the payload data with NOTIFY or INDICATE it will only send up to the 23 - 3 (overhead) bytes. The rest is NOT send UNLESS the MTU size has been changed. 

During BLE.begin() the MAXMTU size is updated from the 23 based on the local HCI-layer. In the case of Apollo3 the MAXMTU is set at 242 bytes.

The MTU size to use can be renegotiated after connection by both peripheral and central, up to the local MAXMTU on either side. I have not seen an ArduinoBLE API for performing a request itself, but the ATT code does have the capability as well as the capability to respond to a change in MTU request from remote. Not sure in your case that is happening from remote, but increasing the MTU size will increase the throughput. In my test with byte_one the payload was 4 bytes (1 data + 3 overhead), with using byte_220 the payload of 23 bytes (20 data + 3 overhead).

Another point is the speed of your remote device. The HCI layer on the Apollo3 can have a maximum of 4 packets pending. If a remote device needs more time to handle the receiving of the packets, the HCI sending routine will wait and loop until a packet/slot becomes available. 

regards,
Paul

paulvha, 2021-12-23:
After a number of changes, I got PythonNUS to work on Ubuntu with the byte_220 sketch as peripheral. I can confirm that PythonNUS is requesting to increase the MTU to 517, but as the maximum on Apollo3 is 242, it is set for 242. That is enough for a 220-bytes payload. Still, the maximum ACL packet is 31 ( 27 payload + 4 BLE overhead) according to BLE specifications. The payload includes other overhead and as such 8 x 31 bytes + 1x 24 bytes are sent for 220 data bytes. 

I have created a "central" sketch to handle the byte-220 simulated peripheral output on an Artemis EDGE, while the peripheral is on an ATP and increased the MTU to 242. It is going with the speed of light. but the central crashes as the MBED-BLE layer buffers do not adjust well with the ArduinoBLE layer buffers. Including a delay of 1 second in sending the next packet on the peripheral / ATP makes it work stable and fast.

I suspect your issue is the handling of 220 bytes by the PythonNUS and not Apollo3. However you also need to look at the total **throughput** : 1000 * 1 byte = 1K bytes,  1000 * 220 bytes = 22k. **YES** that takes longer to transmit and handle, but "only"  3 times as long and not 22 times as long.

regards,
Paul

JacksonLtt, 2021-12-24:
thank you for your advice. 

I think the low latency is because of the handling of PythonNUS.

idea--list, 2022-01-14:
Awesome info @paulvha 

> throughput : 1000 * 1 byte = 1K bytes, 1000 * 220 bytes = 22k

With MbedOS it was really challenging for me to stream 1024 byte/s time sensitive data constantly to my android phone with the Artemis even with 240 byte MTU and thus i thought the maximal throughput was around 1K.

closed 2022-03-17 by Wenn0101